Moddable-OpenSource/moddable #419: Spec. Mismatch:  Promise.race behave differently against the specification

Reinose, 2020-08-11:
### Environment
```
# uname -a
Linux reinose-ubuntu 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

# xs -v
XS 10.3.0
```

### Bug description
```JavaScript
'use strict'
async function x ( y ) { y ( 42 ) };
Promise . race ( { [ Symbol . iterator ] : async function ( ) { await x ( ); } } ) ;
throw Error(42);
```
The above JavaScript code does not produce an exception in xs.

However, according to ECMA262 specification([http://ecma-international.org/ecma-262/11.0/](http://ecma-international.org/ecma-262/11.0/)), the code should throw **Error(42)**

### Details
The execution about **Promise.race** starts from [25.6.4.4 Promise.race](http://ecma-international.org/ecma-262/11.0/#sec-promise.race).
The execution continues to the step 5.
Inside of [25.6.4.4.1 Runtime Semantics: PerformPromiseRace](http://ecma-international.org/ecma-262/11.0/#sec-performpromiserace), `IterationStep(iteratorRecord)` at the step 5.a will return abrupt completion since the iterator calls `x`.
Therefore, the abrupt completion should be propagated at the step 5.c.
Then, the step 6.b in Promise.race is performed and the Promise object should be returned.
There are no reasons to stop the entire execution in this circumstance.
But xs does not throw **Error(42)** after Promise.race finished.

NOTE: The following is the result of Node v14.5.0
```
# node -e "'use strict'
# async function x ( y ) { y ( 42 ) };
# Promise . race ( { [ Symbol . iterator ] : async function ( ) { await x ( ); } } ) ;
# throw Error(42);"
[eval]:4
throw Error(42);
^

Error: 42
    at [eval]:4:7
    at Script.runInThisContext (vm.js:131:18)
    at Object.runInThisContext (vm.js:295:38)
    at Object.<anonymous> ([eval]-wrapper:10:26)
    at Module._compile (internal/modules/cjs/loader.js:1201:30)
    at evalScript (internal/process/execution.js:98:25)
    at internal/main/eval_string.js:23:3
```

Reinose, 2020-08-20:
I've re-inspected this case and noticed that I made a mistake while identifying the bug cause.
```js
'use strict';
async function x ( ) { throw 42 ; } ;
async function y ( ) { await x ( ) ; } ;
y ( ) ;
throw 43 ;
// xst -s ./sample.js
```
This example exits normally without throwing any exceptions.

```js
'use strict';
async function x ( ) { print(1); throw 42 ; print(2); } ;
async function y ( ) { print(3); await x ( ) ; print(4); } ;
y ( ) ;
print(5);
throw 43 ;
print(6);
```
After I add some `print` statements, the result is as follow.
```
3
1
5
```
Other engines such as `V8 (node)`, `QuickJS` and `GraalJS` result in throwing `43` at the end of the above result.

Therefore, the bug might be from evaluating _AwaitExpression_, [14.7.14 Runtime Semantics: Evaluation](http://ecma-international.org/ecma-262/11.0/#sec-async-function-definitions-runtime-semantics-evaluation).

In the case `await UnaryExpression`, step 3 will be executed since _UnaryExpression_ `x ( ) ;` returns a normal completion. (See [14.7.11 Runtime Semantics: EvaluateBody](http://ecma-international.org/ecma-262/11.0/#sec-async-function-definitions-EvaluateBody), [12.3.6 Function Calls](http://ecma-international.org/ecma-262/11.0/#sec-function-calls), and [9.2.1 [[Call]]](http://ecma-international.org/ecma-262/11.0/#sec-ecmascript-function-objects-call-thisargument-argumentslist))
At step 3, **Await** will be called with the promise object which contains a valid **[[reject]]**, since async function `x` throws 42.
Hence, in my opinion, there might be bugs in handling [6.2.3.1 Await](http://ecma-international.org/ecma-262/11.0/#await) when the passed argument is the rejected promise.

phoddie, 2020-08-24:
We found that the code behaves correctly but that the `xst` command line tool did not print the error. That is fixed in the next xs update.

closed 2020-08-24 by phoddie