Moddable-OpenSource/moddable #423: External Interface spec for Moddable demo hardware

mikew67, 2020-08-20:
I note that in issue #221, a request for schematics for your hardware products was denied.  OK, but I do need **External Interface Specs** for your products, especially the Moddable Two.  I intend to build some daughter cards for the MTwo (R6, in the current case), and run it off battery power. Hence, I need these specs, including:

- **6-pin input programming connector**, including its voltage limits and signal characteristics.  For example, if I supply battery power to the +5v input pin, what's the lowest possible voltage?  (Something a single LiPo cell can supply, 3.7-4.1v.). Presuming the internals of the board are all running at 3.3v (LCD, too?), can I supply a filtered 3.3v to some internal point.

- **20-pin external connector**,  including the same info.  Suspect most of the signal pins are direct connects to the ESP32 pins, so I can take the electrical specs from the ESP32 datasheet, but there are a few additional questions, e.g., what's the current capability of the 3.3v output?  How is it filtered?  Is there any significant capacitance on the SCL and SDA lines (need to know when I'm pushing I2C interfaces at the chip's max speed).

- Part number for the mating 6-pin and 20-pin connectors would be useful, too.

- Anything else you can add about the HW and electrical specs.

Thanks,

Mike

chris-moddable, 2020-08-23:
More information on the power and pins of the Moddable Two below. Let us know if this helps in your development.

# Moddable Two Pin Information 
## 6-pin Programming header
| Name | I/O  | Description |
|:------------- |:---------------:| :-------------|
| RTS | I/O | Connects to auto programming and reset circuit |
| DTR | I/O | Connects to auto programming and reset circuit |
| TX | I/O | Connects to ESP32 GPIO-01 |
| RX | I/O | Connects to ESP32 GPIO-03 |
| 5v | I | 5v input connects to LM1117-3.3 voltage regulator and 5v pin on 16-pin external connector |
| GND | GND | Connects to GND |

## 16-pin External header
| Name| I/O  | Description |
|:------------- |:---------------:| :-------------|
| SDA-GP21 | I/O | Connects to ESP32 GPIO21 (standard SDA, no external pull up resistor) |
| SCL-GP22 | I/O | Connects to ESP32 GPIO22 (standard SCL, no external pull up resistor) |
| GND      | GND | Connects to GND |
| 3.3v     | I/O | 3.3v input and output. Connects to ESP32 3.3v input and other 3.3v IC's. Regulated output power if board is 5v powered via micro USB, VIN external connector or programming connector. |
| GP13     | I/O | Connects to ESP32 GPIO13 (Also MOSI for display) |
| GP12     | I/O | Connects to ESP32 GPIO12 |
| GP14     | I/O | Connects to ESP32 GPIO14 |
| GP32     | I/O | Connects to ESP32 GPIO32 |
| GP33     | I/O | Connects to ESP32 GPIO33 |
| GP35     | I/O | Connects to ESP32 GPIO35 |
| GP27     | I/O | Connects to ESP32 GPIO27 |
| GP26     | I/O | Connects to ESP32 GPIO26 |
| VIN      | I/O | VIN is a 5v pin that can be used to power the Moddable Two. If the board is powered by one of the other 5v inputs this pin can be used for external 5v power. This is an un-regulated pin. Connects to other 5v inputs and LM1117-3.3 voltage regulator. |
| GP17     | I/O | Connects to ESP32 GPIO17 |
| GP16     | I/O | Connects to ESP32 GPIO16 |
| GP0      | I/O | Connects to ESP32 GPIO0 |

## Power

The Moddable Two is a 3.3v device. 5v power is regulated to 3.3v by a LM1117-3.3 voltage regulator (see data sheet for specs). Testing of Moddable Two has been with typical 5v 0.5amp USB source power.  

#### Power can be supplied to the Moddable Two via the following: 

* 5v - Micro USB connector (note the Micro USB cannot be used for programming)
* 5v - Moddable Programmer connector
* 5v - VIN on 16 pin external header 
* 3.3v - 3.3v pin external header 

#### External header output power pins 
* VIN is a direct connection to the 5v input sources
* 3.3v is the output of the LM1117-3.3. Output of the LM1117 is conditioned with a 100uf Tantalum capacitor.

mikew67, 2020-08-24:
Thanks!

Mike

phoddie, 2020-08-24:
We'll migrate this information to the Moddable Two documentation, so it is easy to find.

closed 2020-08-24 by phoddie