Moddable-OpenSource/moddable #428: IMUs on M5Stack Fire?

phoddie, 2020-08-27:
I recently made a [change](https://github.com/Moddable-OpenSource/moddable/commit/683ad866a7b5d56d7981c326c4881f1eb14a8756?branch=683ad866a7b5d56d7981c326c4881f1eb14a8756&diff=split) to the M5Stack Fire so that start-up could proceed if the IMU was not found. This was done because @meganetaaan mentioned that some M5Stack Fire hardware uses a different IMU from the `MPU6050` [currently expected](https://github.com/Moddable-OpenSource/moddable/blob/276e962b1106b798b388663f0f311bc32f917c23/build/devices/esp32/targets/m5stack_fire/setup-target.js#L39).

There were [recent changes](https://github.com/Moddable-OpenSource/moddable/blob/276e962b1106b798b388663f0f311bc32f917c23/build/devices/esp32/targets/m5stick_c/setup-target.js#L126) to the M5Stick-C to detect the IMU (SH200Q or MPU6886). This allows the M5Stick-C to work with either IMU without user intervention.

It seems like we should apply a similar change to the M5Stack Fire. To do that, we would need to know which IMUs it has shipped with. I don't know. @meganetaaan, @wilberforce, @Tiryoh  -- do you know?

wilberforce, 2020-08-27:
https://docs.m5stack.com/#/en/core/fire

BMM150 + MPU6886

So we now have a mpu8666 driver.

The MAG is new:

https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmm150-ds001.pdf

phoddie, 2020-08-28:
Oh good. One more driver to write. I don't have the part, so I can't volunteer. Anyone?

Tiryoh, 2020-08-28:
I don't have an M5Stack Fire, but I looked it up on Switch Science, a M5Stack distributor in Japan.
According to Switch Science, the current M5Stack Fire has BMM150 + MPU6886 or BMM150 + SH200Q.
[image]
https://www.switch-science.com/catalog/3953/

Also, it seems like M5Stack Gray has MPU6886 + BMM150 (or MPU9250).
[image]
https://www.switch-science.com/catalog/3648/

related issue: https://github.com/m5stack/M5Stack/issues/191

closed 2022-10-20 by phoddie