Moddable-OpenSource/moddable #429: Consider adding iterator to Resource

wilberforce, 2020-08-27:
I have a folder with WAV files, and reference them in the manifest with:

```
"resources": {have 
		"*":
			"./wav/*"
	}
```

In main.js the wav files are repeated:
```
let clips = [
	'blaster.maud',
	'obi-wan-hello-there.maud',
	'yoda-do-it.maud',
	'yoda-laugh.maud',
	'wilhelmscream.maud',
	'yoda-do-or-do-not.maud',
	'r2d2-scream.maud'
]
```

It would be useful to have an iterator class like:
https://github.com/Moddable-OpenSource/moddable/blob/4a5f43eabc1ed18c217eeb573215a08b0da0fc0b/modules/files/file/file.js#L42-L56

This would prevent having to declare a list, and just iterate the Resources.

phoddie, 2020-08-27:
You can do this for most cases by adding

phoddie, 2020-08-27:
It is straightforward to implement this for the common cases.  Add a static `name` method to [`Resource`](https://github.com/Moddable-OpenSource/moddable/blob/276e962b1106b798b388663f0f311bc32f917c23/modules/files/resource/Resource.js#L21):

```js
	static name(index) @ "Resource_name";
```

Here's the native implementation:

```c
void Resource_name(xsMachine *the)
{
	extern const char *mcGetResourceName(void* it, int i);
	extern int mcCountResources(void* it);
	int index = xsToInteger(xsArg(0));
	if ((index < 0) || (index >= mcCountResources(NULL)))
		return;

	xsResult = xsString(mcGetResourceName(NULL, index));
}
```

It works like this:

```js
import Resource from "Resource";

for (let i = 0; true; i++) {
	const name = Resource.name(i);
	if (!name) break;
	trace(name, "\n");
}
```

You can build a JavaScript iterator around that in the same way as the `File` module you reference.

The implementation doesn't work when there is an archive. That's an admittedly obscure case at the moment, but one that should be addressed for this to be property integrated.

wilberforce, 2020-11-30:
Thanks @phoddie 

https://github.com/Moddable-OpenSource/moddable/commit/4e98dbba10cc48d332c51fd83b476834afebad59

closed 2020-11-30 by wilberforce