Moddable-OpenSource/moddable #432: Heap buffer overflow at moddable/xs/sources/xsSyntaxical.c:3562

kvenux, 2020-08-31:
## Build environment:
Ubuntu 16.04
gcc 5.4.0
xst version: de64c70 (git hash)
build command:
cd /path/to/moddable/xs/makefiles/lin
make
test command: ./xst poc

## Target device:

Desktop Linux

## POC

[000053.txt](https://github.com/Moddable-OpenSource/moddable/files/5148563/000053.txt)

## Description

Below is the ASAN outputs. Heap buffer overflow at moddable/xs/sources/xsSyntaxical.c:3562

[image]
[image]

phoddie, 2020-08-31:
The provided script may be reduced to the following:

```
function v4 = ()(v6, v7) {
}
```

kvenux, 2020-09-01:
> The provided script may be reduced to the following:
> 
> ```
> function v4 = ()(v6, v7) {
> }
> ```

Great. I have checked. This test case can be trimmed in this way.

closed 2020-09-04 by kvenux