Moddable-OpenSource/moddable #433: Heap buffer overflow at moddable/xs/sources/xsDebug.c:784

kvenux, 2020-08-31:
## Build environment:
Ubuntu 16.04
gcc 5.4.0
xst version: de64c70 (git hash)
build command:
cd /path/to/moddable/xs/makefiles/lin
make
test command: ./xst poc

## Target device:

Desktop Linux

## POC

[000523.txt](https://github.com/Moddable-OpenSource/moddable/files/5148577/000523.txt)

## Description

Below is the ASAN outputs. Heap buffer overflow at /moddable/xs/sources/xsDebug.c:784

=================================================================
==118435==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7fc129d737ea at pc 0x00000062bd16 bp 0x7ffccdfbfe90 sp 0x7ffccdfbfe80
READ of size 2 at 0x7fc129d737ea thread T0
    #0 0x62bd15 in fxDebugThrow /home/keven/Fuzzing/moddable/xs/sources/xsDebug.c:784
    #1 0x441af0 in fxThrowMessage /home/keven/Fuzzing/moddable/xs/sources/xsAPI.c:1257
    #2 0xa526d8 in fxAbort /home/keven/Fuzzing/moddable/xs/tools/xst.c:1378
    #3 0x93fb78 in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:253
    #4 0x4432c0 in fxToString /home/keven/Fuzzing/moddable/xs/sources/xsAPI.c:312
    #5 0x477f02 in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1549
    #6 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #7 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #8 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #9 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #10 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #11 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #12 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #13 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #14 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #15 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
    #16 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #17 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #18 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #19 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #20 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #21 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #22 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #23 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #24 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #25 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
    #26 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #27 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #28 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #29 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #30 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #31 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #32 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #33 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #34 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #35 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
    #36 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #37 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #38 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #39 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #40 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #41 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #42 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #43 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #44 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #45 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
[... 211 lines cut ...]
    #257 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #258 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #259 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #260 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #261 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #262 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #263 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #264 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #265 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
    #266 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #267 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #268 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #269 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336
    #270 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #271 0x9400ae in fxToPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsType.c:268
    #272 0x7fe37f in fxToNumericNumber /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4464
    #273 0x7fe37f in fxToNumericNumberBinary /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:4483
    #274 0x82c555 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:3337
    #275 0x477e6e in fx_Array_prototype_join /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:1546
    #276 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #277 0x468e7c in fx_Array_prototype_toString /home/keven/Fuzzing/moddable/xs/sources/xsArray.c:2333
    #278 0x7ffe39 in fxRunID /home/keven/Fuzzing/moddable/xs/sources/xsRun.c:767
    #279 0x77270c in fx_Object_prototype_toPrimitive /home/keven/Fuzzing/moddable/xs/sources/xsObject.c:336

[image]

phoddie, 2020-09-01:
I'd like to understand how you are using the address-sanitizer here to ensure we are looking at the same thing. The notes at the top suggest the report is based on running an unmodified copy of `xst`. However, in my experience, using address-sanitizer requires changes to build flags.  Would you please provide in the missing detail? Thank you.

kvenux, 2020-09-02:
> I'd like to understand how you are using the address-sanitizer here to ensure we are looking at the same thing. The notes at the top suggest the report is based on running an unmodified copy of [...]

I added the following in xs/makefiles/lin/xst.mk:

diff --git a/xs/makefiles/lin/xst.mk b/xs/makefiles/lin/xst.mk
index 4fb7f05..ebfe992 100644
--- a/xs/makefiles/lin/xst.mk
+++ b/xs/makefiles/lin/xst.mk
@@ -38,9 +38,12 @@ TMP_DIR = $(BUILD_DIR)/tmp/lin/$(GOAL)/$(NAME)

 MACOS_ARCH ?= -arch i386
 MACOS_VERSION_MIN ?= -mmacosx-version-min=10.7
+CC=/path/to/afl-gcc
+CXX=/path/to/afl-g++

 C_OPTIONS = \
        -fno-common \
+       -fsanitize=address \
        -DINCLUDE_XSPLATFORM \
        -DXSPLATFORM=\"xst.h\" \
        -DmxDebug=1 \

Is this unreproducible problem caused by afl-gcc? 

Btw, I have some other similar POCs.

[xs-000404.txt](https://github.com/Moddable-OpenSource/moddable/files/5159473/xs-000404.txt)

phoddie, 2020-09-02:
Running on macOS with ASAN enabled. I cannot reproduce a crash with 000523.txt or xs-000404.txt. I'm not using `alf-gcc`. I don't know if `alf-gcc` explains that or not.

FWIW - I am able to reproduce the problem in #431 - but only if the debugger is running. In "build environment" above, you don't note whether xsbug is running or not.

kvenux, 2020-09-03:
@phoddie I have reproduced 000523.txt on MacOS.

Platform:
ProductName:    Mac OS X
ProductVersion: 10.14.6
BuildVersion:   18G6020

gcc version:
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

compile setting:
I added the following in xs/makefiles/lin/xst.mk
C_OPTIONS += -fsanitize=address
LINK_OPTIONS += -fsanitize=address

test cmd:
./build/bin/mac/release/xst 000523.txt

ASAN outputs:

=================================================================
==64588==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x00010d9797ea at pc 0x0001079bb261 bp 0x7ffee83059d0 sp 0x7ffee83059c8
READ of size 2 at 0x00010d9797ea thread T0
    #0 0x1079bb260 in fxDebugThrow (xst:x86_64+0x10013d260)
    #1 0x107880f12 in fxThrowMessage (xst:x86_64+0x100002f12)
    #2 0x107c007a5 in fxAbort (xst:x86_64+0x1003827a5)
    #3 0x107b620bd in fxToPrimitive (xst:x86_64+0x1002e40bd)
    #4 0x107881a8f in fxToString (xst:x86_64+0x100003a8f)
    #5 0x10789e9d2 in fx_Array_prototype_join (xst:x86_64+0x1000209d2)
    #6 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #7 0x1078abac7 in fx_Array_prototype_toString (xst:x86_64+0x10002dac7)
    #8 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #9 0x107a7f828 in fx_Object_prototype_toPrimitive (xst:x86_64+0x100201828)
    #10 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #11 0x107b6251c in fxToPrimitive (xst:x86_64+0x1002e451c)
    #12 0x107ad2ecc in fxRunID (xst:x86_64+0x100254ecc)
    #13 0x10789e979 in fx_Array_prototype_join (xst:x86_64+0x100020979)
    #14 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #15 0x1078abac7 in fx_Array_prototype_toString (xst:x86_64+0x10002dac7)
    #16 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #17 0x107a7f828 in fx_Object_prototype_toPrimitive (xst:x86_64+0x100201828)
    #18 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #19 0x107b6251c in fxToPrimitive (xst:x86_64+0x1002e451c)
    #20 0x107ad2ecc in fxRunID (xst:x86_64+0x100254ecc)
    #21 0x10789e979 in fx_Array_prototype_join (xst:x86_64+0x100020979)
    #22 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #23 0x1078abac7 in fx_Array_prototype_toString (xst:x86_64+0x10002dac7)
    #24 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #25 0x107a7f828 in fx_Object_prototype_toPrimitive (xst:x86_64+0x100201828)
    #26 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #27 0x107b6251c in fxToPrimitive (xst:x86_64+0x1002e451c)
    #28 0x107ad2ecc in fxRunID (xst:x86_64+0x100254ecc)
    #29 0x10789e979 in fx_Array_prototype_join (xst:x86_64+0x100020979)
    #30 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #31 0x1078abac7 in fx_Array_prototype_toString (xst:x86_64+0x10002dac7)
    #32 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #33 0x107a7f828 in fx_Object_prototype_toPrimitive (xst:x86_64+0x100201828)
    #34 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #35 0x107b6251c in fxToPrimitive (xst:x86_64+0x1002e451c)
    #36 0x107ad2ecc in fxRunID (xst:x86_64+0x100254ecc)
    #37 0x10789e979 in fx_Array_prototype_join (xst:x86_64+0x100020979)
    #38 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #39 0x1078abac7 in fx_Array_prototype_toString (xst:x86_64+0x10002dac7)
    #40 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
    #41 0x107a7f828 in fx_Object_prototype_toPrimitive (xst:x86_64+0x100201828)
    #42 0x107adce54 in fxRunID (xst:x86_64+0x10025ee54)
[... 231 lines cut ...]
  0x100021b2f310: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100021b2f320: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100021b2f330: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100021b2f340: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==64588==ABORTING

phoddie, 2020-09-03:
Thank you very much for trying this on macOS and for providing the additional build details.

I'm not seeing the problem, but... now I'm running a more recent version of the code. We just pushed an update that contains fixes for a couple issues you reported that we could reproduce. Would you retry this one to see if it was resolved as part of that? If not, we'll need to sort out why I'm still unable to reproduce this one.

kvenux, 2020-09-04:
@phoddie It's my pleasure. Hunting for bugs is fun

I have tried on the latest version 5639abb2. 000523.txt can still trigger a crash but turns out to be another one (stack overflow). That's weird.
Can you check this again? Any contributor can check this one too?

xst version: 5639abb2

Platform:
ProductName: Mac OS X
ProductVersion: 10.14.6
BuildVersion: 18G6020

gcc version:
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/include/c++/4.2.1
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

compile setting:
I added the following in xs/makefiles/lin/xst.mk
C_OPTIONS += -fsanitize=address
LINK_OPTIONS += -fsanitize=address

test cmd:
./build/bin/mac/release/xst 000523.txt

ASAN outputs:

AddressSanitizer:DEADLYSIGNAL
=================================================================
==79835==ERROR: AddressSanitizer: stack-overflow on address 0x7ffee1458f98 (pc 0x00010e251f42 bp 0x7ffee1468090 sp 0x7ffee1458fa0 T0)
    #0 0x10e251f41 in fxRunID xsRun.c:767
    #1 0x10e348e16 in fxToPrimitive xsType.c:268
    #2 0x10dfaaf65 in fxToString xsAPI.c:312
    #3 0x10dfdf247 in fx_Array_prototype_join xsArray.c:1549
    #4 0x10e251f43 in fxRunID xsRun.c:767
    #5 0x10dff8dfc in fx_Array_prototype_toString xsArray.c:2333
    #6 0x10e251f43 in fxRunID xsRun.c:767
    #7 0x10e1cb46e in fx_Object_prototype_toPrimitive xsObject.c:336
    #8 0x10e251f43 in fxRunID xsRun.c:767
    #9 0x10e348e16 in fxToPrimitive xsType.c:268
    #10 0x10e2c2e8f in fxToNumericNumber xsRun.c:4464
    #11 0x10e2c2b07 in fxToNumericNumberBinary xsRun.c:4483
    #12 0x10e2a3eb8 in fxRunID xsRun.c:3337
    #13 0x10dfb3724 in fxGetAll xsAPI.c:891
    #14 0x10dfb3b4e in fxGetIndex xsAPI.c:919
    #15 0x10dfdf067 in fx_Array_prototype_join xsArray.c:1546
    #16 0x10e251f43 in fxRunID xsRun.c:767
    #17 0x10dff8dfc in fx_Array_prototype_toString xsArray.c:2333
    #18 0x10e251f43 in fxRunID xsRun.c:767
    #19 0x10e1cb46e in fx_Object_prototype_toPrimitive xsObject.c:336
    #20 0x10e251f43 in fxRunID xsRun.c:767
    #21 0x10e348e16 in fxToPrimitive xsType.c:268
    #22 0x10e2c2e8f in fxToNumericNumber xsRun.c:4464
    #23 0x10e2c2b07 in fxToNumericNumberBinary xsRun.c:4483
    #24 0x10e2a3eb8 in fxRunID xsRun.c:3337
    #25 0x10dfb3724 in fxGetAll xsAPI.c:891
    #26 0x10dfb3b4e in fxGetIndex xsAPI.c:919
    #27 0x10dfdf067 in fx_Array_prototype_join xsArray.c:1546
    #28 0x10e251f43 in fxRunID xsRun.c:767
    #29 0x10dff8dfc in fx_Array_prototype_toString xsArray.c:2333
    #30 0x10e251f43 in fxRunID xsRun.c:767
    #31 0x10e1cb46e in fx_Object_prototype_toPrimitive xsObject.c:336
    #32 0x10e251f43 in fxRunID xsRun.c:767
    #33 0x10e348e16 in fxToPrimitive xsType.c:268
    #34 0x10e2c2e8f in fxToNumericNumber xsRun.c:4464
    #35 0x10e2c2b07 in fxToNumericNumberBinary xsRun.c:4483
    #36 0x10e2a3eb8 in fxRunID xsRun.c:3337
    #37 0x10dfb3724 in fxGetAll xsAPI.c:891
[... 196 lines cut ...]
    #234 0x10e251f43 in fxRunID xsRun.c:767
    #235 0x10e1cb46e in fx_Object_prototype_toPrimitive xsObject.c:336
    #236 0x10e251f43 in fxRunID xsRun.c:767
    #237 0x10e348e16 in fxToPrimitive xsType.c:268
    #238 0x10e2c2e8f in fxToNumericNumber xsRun.c:4464
    #239 0x10e2c2b07 in fxToNumericNumberBinary xsRun.c:4483
    #240 0x10e2a3eb8 in fxRunID xsRun.c:3337
    #241 0x10dfb3724 in fxGetAll xsAPI.c:891
    #242 0x10dfb3b4e in fxGetIndex xsAPI.c:919
    #243 0x10dfdf067 in fx_Array_prototype_join xsArray.c:1546
    #244 0x10e251f43 in fxRunID xsRun.c:767
    #245 0x10dff8dfc in fx_Array_prototype_toString xsArray.c:2333
    #246 0x10e251f43 in fxRunID xsRun.c:767
    #247 0x10e1cb46e in fx_Object_prototype_toPrimitive xsObject.c:336
    #248 0x10e251f43 in fxRunID xsRun.c:767
    #249 0x10e348e16 in fxToPrimitive xsType.c:268
    #250 0x10e2c2e8f in fxToNumericNumber xsRun.c:4464
    #251 0x10e2c2b07 in fxToNumericNumberBinary xsRun.c:4483
    #252 0x10e2a3eb8 in fxRunID xsRun.c:3337
    #253 0x10dfb3724 in fxGetAll xsAPI.c:891
    #254 0x10dfb3b4e in fxGetIndex xsAPI.c:919
    #255 0x10dfdf067 in fx_Array_prototype_join xsArray.c:1546

SUMMARY: AddressSanitizer: stack-overflow xsRun.c:767 in fxRunID
==79835==ABORTING

phoddie, 2020-09-04:
Thanks for working through this.

That's a native stack overflow. On macOS, I don't see that. I see the XS stack overflowi, and XS cleanly exit the virtual machine.

However, if the XS stack is increased, then something similar to your report above happens. To increase the XS stack increase `stackCount` here:

https://github.com/Moddable-OpenSource/moddable/blob/5639abb24b6d725554969dc0be5822edb54a4a08/xs/tools/xst.c#L284-L286

by 100x:

```c
	4096 * 100, 		/* stackCount */
```

With that change, here's the output:

```
> xst ~/test.js 
AddressSanitizer:DEADLYSIGNAL
=================================================================
==6854==ERROR: AddressSanitizer: stack-overflow on address 0x7ffeec86ed48 (pc 0x000102deae36 bp 0x7ffeec86f5c0 sp 0x7ffeec86ed50 T0)
    #0 0x102deae35 in wrap_memmove (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x1de35)
    #1 0x7fff6d00512d in __sfvwrite (libsystem_c.dylib:x86_64+0x3d12d)
    #2 0x7fff6d00e636 in __vfprintf (libsystem_c.dylib:x86_64+0x46636)
    #3 0x7fff6d0331c2 in __v2printf (libsystem_c.dylib:x86_64+0x6b1c2)
    #4 0x7fff6d018b02 in _vsnprintf (libsystem_c.dylib:x86_64+0x50b02)
    #5 0x7fff6d018ba9 in vsnprintf (libsystem_c.dylib:x86_64+0x50ba9)
    #6 0x102df654a in wrap_vsnprintf (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x2954a)
    #7 0x102df74d5 in wrap_snprintf (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x2a4d5)
    #8 0x102c7aae0 in fxIntegerToString (xst:x86_64+0x1000e9ae0)
    #9 0x102b9394a in fxToString (xst:x86_64+0x10000294a)
    #10 0x102b9c87b in fx_Array_prototype_join (xst:x86_64+0x10000b87b)
    #11 0x102c32137 in fxRunID (xst:x86_64+0x1000a1137)
    #12 0x102ba0c98 in fx_Array_prototype_toString (xst:x86_64+0x10000fc98)
    #13 0x102c32137 in fxRunID (xst:x86_64+0x1000a1137)
...
    #255 0x102c32137 in fxRunID (xst:x86_64+0x1000a1137)

SUMMARY: AddressSanitizer: stack-overflow (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x1de35) in wrap_memmove
==6854==ABORTING
Abort trap: 6

```

Without ASAN enabled, the output is just:

```
Segmentation fault: 11
```

I don't think there's anything here to fix. XS isn't going to check the stack on every function call. The host OS catches the stack overflow and kills the process.

kvenux, 2020-09-04:
Confirmed. This is a naive stack overflow. Not a bug.

closed 2020-09-04 by kvenux