Moddable-OpenSource/moddable #434: M5Stack LCD is changed to ILI9342 from ILI9341. Display color is inverted.

stc1988, 2020-08-31:
I have M5Stack Gray that LCD is changed to ILI9342.
It displayed fine with ILI9341 driver, but color is inverted

https://twitter.com/stc1988/status/1280089696899481601?s=20

I created temploraly fix to ILI9341 driver.It works fine.

`$(MODULES)/drivers/ili9341/modIli9341.c`

```
void ili9341Init(spiDisplay sd)
{
        // skip

        ili9341Command(sd, 0x21, NULL, 0); // add temporaary fix. color invert command.
        sd->firstFrame = true;
}
```

M5Stack library detects lcd version and sends inversion command.It looks so good solution.
https://github.com/m5stack/M5Stack/blob/f10648586cc4324d844cfd9505cfa47d3d61b7ca/src/utility/In_eSPI.cpp#L438-L440

I referred to the link below.
https://community.m5stack.com/topic/1736/m5stack-display-shows-everything-negative-or-inverted/2

phoddie, 2020-08-31:
Thank you for the report and background.

Let's see if there's a clean solution to this, one that does not require board specific changes to the native driver that is shared across many devices.

phoddie, 2020-08-31:
For M5Stack,I added a wrapper around the `ILI9341` driver. It extends the constructor to be able to optionally invert thee screen.

https://github.com/Moddable-OpenSource/moddable/blob/e9b1ea4a4f09782b9af3526063968657483bfbdd/build/devices/esp32/targets/m5stack/screen.js#L4-L11

I don't have an M5Stack with the revised screen, so I'm not sure which pin to read at line 8 to determine the `lcd_version`. In the M5Stack library you link to above, they show it reading from `TFT_RST` but i don't know which GPIO that maps to. @stc1988, can you help find that number?

FYI - to support this, I added a `command` method to the ILI9341 class to allow scripts to send commands directly. The new call is briefly documented [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/drivers/ili9341/ili9341.md#command-api).

stc1988, 2020-09-01:
@phoddie 
Thank you for your check.

> In the M5Stack library you link to above, they show it reading from TFT_RST but i don't know which GPIO that maps to. @stc1988, can you help find that number?

I found  that `TFT_RST` is GPIO 33. from [here](https://github.com/m5stack/M5Stack/blob/b94cca295c0de8151e801bee70b00ad1fba72a42/src/utility/In_eSPI_Setup.h#L186)

I use `Digital.read(33)` and it works as expected.

phoddie, 2020-09-01:
Cool. Thanks for the info and for checking that it works. I'll update the code to use GPIO 33.

closed 2020-09-01 by phoddie