Moddable-OpenSource/moddable #435: How to get the RAM and ROM usage of a program running on XS?

paseaf, 2020-09-02:
I'm doing some experiments on Moddable Two and trying to measure how much RAM and ROM space is required for different programs (include XS itself).

My naive assumption is that `RAM_usage = chunk_used + slot_used + stack_used` (read from `xsbug`), is that correct?
As for ROM, I haven't found any good way to measure that. Do you have any suggestion?

Any help will be appreciated!

phoddie, 2020-09-02:
Regarding RAM usage, the information displayed in xsbug is a very good first estimate. As you note, it shows the stack, chunk, and slot use which is usually the majority of memory allocated by a running script. There are exceptions. For example, `SharedArrayBuffer` is allocated from the system heap (`ArrayBuffer` is in the chunk heap). Of course, host services use system memory (lwip socket, FreeRTOS tasks, etc).

The information displayed in the xsbug Instruments pane is also available programmatically using the `Instrumentation` module. The API is documented [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/base/base.md#instrumentation. The [xsuse example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/base/instrumentation/xsuse/main.js) shows how to measure the memory use of various objects. (Note: the instrumentation indices in the app are incorrect for ESP32 - in the calls to `Instrumentation.get` the index should be increased by 3.)

Regarding ROM use, the link map is the definitive way to see the contribution of each source file to the firmware image. The build always generates a link map. For the esp32/moddable_two build of xsuse, the link map is  located at `⁨$MODDABLE// ⁨build⁩/ ⁨bin⁩/ ⁨esp32⁩/ ⁨debug⁩/moddable_two/⁨xsuse⁩/xs_esp32.map`. The relevant information begins after the text `Linker script and memory map`.

paseaf, 2020-09-02:
Understood! Thanks a lot for your help! :D 

P.S., maybe a small typo in linker map's path: I found it under `$MODDABLE/build/bin/esp32/moddable_two/debug/xsuse/xs_esp32.map` instead of `⁩/ ⁨debug⁩/moddable_two` :)

closed 2020-09-02 by phoddie

paseaf, 2020-09-19:
I did some experiments and now have some more questions.
1. Does XS provide APIs for measuring the RAM usage of the XS runtime and an application running on XS? (Or is it better to look for FreeRTOS or ESP32 APIs?)

2. Why is the available Chunk + Slot + Stack size on Moddable Two only around 14 KiB while ESP32 has 520 KiB RAM in total? Is the size configurable?
[image]

3. As for ROM, does XS provide any APIs for getting the used and available space in the `"factory"` partition?

Any help will be appreciated!

phoddie, 2020-09-19:
> Does XS provide APIs for measuring the RAM usage of the XS runtime and an application running on XS? (Or is it better to look for FreeRTOS or ESP32 APIs?)

The Instruments panel in xsbug shows the total free system memory as well as detailed use of XS itself (chunks, slots, and stack). It is not clear what you are expecting beyond that.

> Why is the available Chunk + Slot + Stack size on Moddable Two only around 14 KiB while ESP32 has 520 KiB RAM in total? Is the size configurable?

Instrumentation shows the current and peak use for chunks, slots, and the stack. These are all contained within the static allocation for the virtual machine specified in the manifest. The default is 32KB, as specified in manifest_base.json, and may be overridden in a project's manifest.

https://github.com/Moddable-OpenSource/moddable/blob/54d8e94710ed6f8c310f046c4fbc83c967ccd346/examples/manifest_base.json#L8-L9

Because XS allocates memory incrementally within the static allocation, the sum of the stack, chunks, and heap may be less than the static allocation size. Their sum, however, will never exceed the static allocation size.

N.B. While the ESP32 datasheet indicates that there is 520 KB of memory, that memory is not all general purpose and is not in a contiguous memory range. Practically speaking, the available RAM is closer to half that.

> As for ROM, does XS provide any APIs for getting the used and available space in the "factory" partition?

The `factory` partition is where the application's firmware image is installed. When you use `mcconfig` to install a firmware image, it invokes `esptool.py` which outputs something like the following.

```
Serial port /dev/cu.usbserial-BD5A355789
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 50:02:91:8f:10:14
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Configuring flash size...
Compressed 16336 bytes to 11560...
Wrote 16336 bytes (11560 compressed) at 0x00001000 in 0.1 seconds (effective 985.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 6258.2 kbit/s)...
Hash of data verified.
Compressed 757232 bytes to 477499...
Wrote 757232 bytes (477499 compressed) at 0x00010000 in 7.5 seconds (effective 807.2 kbit/s)...
```
The last piece installed is the firmware image. In the above, you can see if is `757232` bytes. That is the a amount of space used in the factory partition.

paseaf, 2020-09-20:
Thanks a lot for your answer! That was very helpful!!

> The Instruments panel in xsbug shows the total free system memory as well as detailed use of XS itself (chunks, slots, and stack). It is not clear what you are expecting beyond that.

I was looking for some function in XS that is similar to `esp_get_free_heap_size()`, but I didn't realize `System bytes free` in xsbug is using `esp_get_free_heap_size()`. Reading from the xsbug should be good enough!

---
I have made some notes for each partition in the default partition table at [build/devices/esp32/xsProj/partitions.csv](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/xsProj/partitions.csv).
1. Could you tell me if I understand the `factory` and `xs` partitions correctly?
2. What is the `settings` partition used for? For preferences data?

| Name         | Type | SubType | Offset   | Size  | Note                                                         |
| :----------- | :--- | :------ | :------- | :---- | ------------------------------------------------------------ |
| #secure_boot |      |         | 0x0000 | 4K  | [secure boot etc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#first-stage-bootloader). |
| #boot        | data | 0       | 0x1000   | 4K    | Stores the [second stage bootloader](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/general-notes.html#second-stage-bootloader) |
| #reserved    | 0    | 0       | 0x2000   | 24K   | ESP System reserved                                          |
| #partition   | data | 0       | 0x8000   | 4K    | Stores this partition table                                   |
| nvs          | data | nvs     | 0x9000   | 24K   | Stores [Non-volatile storage library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html) |
| phy_init     | data | phy     | 0xF000   | 4K    | Stores [PHY](https://en.m.wikipedia.org/wiki/PHY) initialization data. [It is optional](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html?highlight=phy_init#subtype). |
| factory      | app  | factory | 0x10000  | 3648K | Stores the host (a.k.a base application) flashed with `mcconfig`. |
| xs           | 0x40 | 1       | 0x3A0000 | 256K  | Stores the [Mods](https://github.com/Moddable-OpenSource/moddable/blob/de64c7089d6318661f69704c8da7a36c94c39e38/documentation/xs/mods.md#esp32) flashed with `mcrun`. |
| settings     | app  | 1       | 0x3E0000 | 64K   | ?                                                            |
| storage      | data | spiffs  | 0x3F0000 | 64K   | Used for SPIFFS.                                             |

phoddie, 2020-09-20:
That looks about right. Most of the partitions are defined by the ESP-IDF, except for `xs`, `storage`, and `spiffs`. The `settings` partition appears to be unused.

paseaf, 2020-09-21:
Thanks a lot!!!