Moddable-OpenSource/moddable #437: NMAKE fails to build TypeScript example

JanVanBraeckel, 2020-09-03:
**Build environment:** Windows
**Target device:** NodeMCU ESP8266.

**Description**
NMAKE fails when trying to build the helloworld example in **TypeScript**

Console output:
```
mcconfig -m -d -p esp
NMAKE : fatal error U1073: don't know how to make '%MODDABLE%\build\tmp\esp\debug\helloworld\modules\main.js'
Stop.
```

main.ts:
```
debugger;

let message = "Hello, world - TypeScript";
trace(message + "\n");

```

manifest.json:
```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_typings.json"
	],
	"modules": {
		"*": [
			"./main"
		]
	},
}
```

(Moddable env variable substituted for privacy)

**Steps to Reproduce**
1. Rename main.js to main.ts in the helloworld example
2. Add type declaration files to the include array
3. Run the build using mcconfig -m -d -p esp

**Expected behavior**
The build does not fail

andycarle, 2020-09-04:
Thank you for the report. I have confirmed the build error on Windows and we are investigating a fix.

andycarle, 2020-09-05:
@JanVanBraeckel We have pushed changes to fix support for TypeScript in Windows builds. You will need to pull the changes and then re-build the Moddable SDK tools to get the fixes. 

Please give it a shot and let me know how it goes. Thank you again for the report!

&nbsp;- Andy

JanVanBraeckel, 2020-09-05:
@andycarle Works perfectly now without errors, thank you for looking into this and props for the quick resolution!

closed 2020-09-05 by phoddie