Moddable-OpenSource/moddable #438: the 'mcconfig' command error

Hebiao19, 2020-09-03:
The 'mcconfig' command errors threw by Windows terminal when i input 'mcconfig -d -m -p win' for the example 'helloworld'

Here is my system environment variable:
C:\%MODDABLE%\build\bin\win\release

Check many possibilities, but still failed!

Expect your help, Thanks very much!

andycarle, 2020-09-04:
Hi @Hebiao19, let's see if we can get you up and running here. 

To build with the Moddable SDK on Windows, you need to make two changes to environment variables: 

- First, you must define MODDABLE and set it to the path where you cloned the Moddable SDK.
- Second, you add %MODDABLE%\build\bin\win\release to your PATH.

It looks like you may have a typo in your PATH variable. You have `C:%MODDABLE%\build\bin\win\release` when it should just be `%MODDABLE%\build\bin\win\release`. The `C:` part is already in the MODDABLE variable.

Please try that change and let me know how it goes.

Cheers!
&nbsp;- Andy

Hebiao19, 2020-09-04:
> Hi @Hebiao19, let's see if we can get you up and running here.
>
>
>
> To build with the Moddable SDK on Windows, you need to make two changes to environment variables:
>
>
>
> - First, you must define [...]

Hello Andy!

     Thanks so much for your time on my trouble! Unfortunately that I have tried as you told, but the same error still exist!

andycarle, 2020-09-04:
@Hebiao19 Could you dump the contents of your environment variables in the console and paste them here? (Feel free to edit out any sensitive information.) That way I can take a look at your environment to see if I can find the problem.

You can print out all of the environment variables by using the `set` command (with no arguments) at the command prompt.

Hebiao19, 2020-09-04:
> @Hebiao19 Could you dump the contents of your environment variables in the console and paste them here? (Feel free to edit out any sensitive information.) That way I can take a look at your [...]

@andycarle Ok! I will do that after i finish my business trip.

phoddie, 2020-09-23:
Is there an update to this? If not, we'll close this out.

Hebiao19, 2020-09-23:
Hello, Phoddie, Sorry for late response! You can close the topic now. Thank you!

closed 2020-09-23 by phoddie