Moddable-OpenSource/moddable #439: Is there a higher resolution timer

LaurenceMason, 2020-09-04:
I am looking for a way to time more accurately than 1 ms. Are any of the 64 bit general purpose hardware times available and can the C interface be used to access them?

phoddie, 2020-09-04:
The Moddable SDK provides time in milliseconds two ways:

- Via the JavaScript `Date` object
- Via `Time.ticks`

It does not provide higher resolution time through any currently built-in function. Most host platforms provide higher resolution time in some way, such as microseconds. The ESP32 provides microseconds with`esp_timer_get_time`. The design question is how to make microseconds available through a JavaScript API. Adding microseconds to JavaScript `Date` would be non-standard, but adding `Time.microseconds` is no problem.

Rather than modifying the existing `Time` module, we can add a `microseconds` getter function. Here's a simple JavaScript module, `microseconds.js` that adds the microseconds getter to the `Time` object.

```js
import Time from "time";

function microseconds() @ "xs_time_microseconds";

Time.__defineGetter__("microseconds", microseconds);
```

The implementation of the `microseconds` getter is in a C function named `xs_time_microseconds`. Here's the corresponding `microseconds.c`:

```c
#include "xsmc.h"
#include "esp_timer.h"

void xs_time_microseconds(xsMachine *the)
{
	xsNumberValue microseconds = (xsNumberValue)esp_timer_get_time();
	xsmcSetNumber(xsResult, microseconds);
}
```
Store `microseconds.js` and `microseconds.c` in a new directory at `$MODDABLE/modules/base/microseconds`. Then, add the microseconds module to the `helloworld` manifest. 

```json
{
	"include": "$(MODDABLE)/examples/manifest_base.json",
	"modules": {
		"*": [
			"./main",
			"$(MODDABLE)/modules/base/microseconds/*"
		]
	},
	"preload": "microseconds"
}
```
> Note: It is **essential** that the `microseconds` module is preloaded so that it can add the `microseconds` getter to the `Time` class at build time.

You need to make one small change to `Time.js` -- remove the call to `Object.freeze`. This is obsolete, as it is done automatically by the XS linker `xsl`. If you leave this in, the call to `__defineGetter__` will fail. (We'll take this call out of the Moddable SDK sources shortly).

https://github.com/Moddable-OpenSource/moddable/blob/5639abb24b6d725554969dc0be5822edb54a4a08/modules/base/time/time.js#L40

That's all the set-up. To try it out, change the `helloworld` script to use `Time.microseconds`, for example:

```js
import Time from "time"

for (let i = 0; i < 10; i++)
	trace(Time.microseconds + "\n");
```

You could, of course, make `microseconds` a separate module instead of patching it into `Time`. However, it seems natural to extend `Time`. We could also just build `microseconds` into `Time`, but that would include the code for `microseconds` in projects that don't use it.

If people feel this makes sense, we can look at adding it to the Moddable SDK.

LaurenceMason, 2020-09-04:
Thanks for posting this. I learned a lot from it.

My vote would be to add it to the Time assuming that it can work for all the supported devices.

phoddie, 2020-09-09:
I [added](https://github.com/Moddable-OpenSource/moddable/commit/db94611e747fa942b6d1d5167471b880fff3fc94) the microseconds modules as above to the current build. It is implemented for ESP32, ESP8266, and macOS. Perhaps others can help with the remaining platforms.

Down the road, this could migrate into the default `Time` object, rather than being an  option.

closed 2020-09-09 by phoddie