Moddable-OpenSource/moddable #443: Support ESP32-S2

sfranzyshen, 2020-09-05:
Are there plans to support eh ESP32-S2 platform?

phoddie, 2020-09-06:
Good question. I assume we will support it eventually. It should be straightforward. A couple questions for you, since you brought up the subject. ;)

- Is there a common reference development board at this point?
- What specifically about the ESP32-S2 is interesting for you that isn't already in the ESP32?

Thanks.

sfranzyshen, 2020-09-06:
> * Is there a common reference development board at this point?

There are two modules the ESP32-S2-WROOM and ESP32-S2-WROVER with two development boards ... the ESP32-S2-Saola-1 and the ESP32-S2-Kaluga Dev Kit 

> * What specifically about the ESP32-S2 is interesting for you that isn't already in the ESP32?

It’s lower cost than the ESP32 with lower power consumption, Wifi TOF (time of flight), ULP (RISC-V), USB-OTG, real LCD and camera support ... The ESP32-S2 is not an ESP32 killer. Instead, it’s more an ESP8266 killer ... with the LX7 cpu ... this is part of the future espressif product line ... ESP32-S3 (dual-core bluetooth), ESP32-C3, and finally the ESP64 ... the 64bit version

phoddie, 2020-09-06:
Thanks for the info!

It looks like the biggest challenge in supporting the ESP32-S2 is that it requires a minimum of ESP-IDF 4.2. We are currently on 3.3.x. That's just work to solve, but I don' have a sense if the effort yet. Maybe you do?

sfranzyshen, 2020-09-07:
@phoddie thank you for your response ... I wish I had more experience with the differences between eps-idf versions ... I've been watching the changes with another project moving from 3.3 to 4.0 and can see that there are quite a few changes ... Here's a [change log from the 4.0-rc](https://github.com/espressif/esp-idf/releases/tag/v4.0-rc) that has a list of the changes between v3.3 and v4.0 ... maybe this could help to create a plan ... I guess it wouldn't make sense to support both versions ... and a new LTS is still like 18 months away ...

phoddie, 2020-09-07:
@sfranzyshen - Thanks for the notes. In reviewing the change log, it doesn't look too horrible. But... it is tough to estimate without just jumping into the work.

Maybe someone else has experience to share about the transition.

Anyway... getting back to your original question - we'd be happy to support the ESP32-S2. It isn't currently a priority. If someone is inspired to take the lead on that, that would be entirely welcome.

nickbedbury, 2020-12-18:
I would second the ESP32-S2 request.  I'm guessing folks entering into ESP development for LCDs are likely to purchase the Kaluga dev kit going forward (as I recently did a week ago).  

Under ESP-IDF 4.3x, it does appear that support for ESP32-S2 is still in flux though...  That could mean extra work if trying to pull it in too early.  Perhaps wait until 4.3 (with a big ol' note about supporting ESP32-S2) is officially released?

phoddie, 2020-12-18:
Hey, @nicholasareed. Yes, we are looking into this. No news to share yet, but perhaps soon.

phoddie, 2021-01-15:
Good news! ESP32-S2 support is now available in a branch. This includes support for ESP-IDF 4.2. Details are on our [blog](https://blog.moddable.com/blog/espidf42). Please give it a try. We intend to merge the branch with main in early February.

phoddie, 2021-02-10:
ESP32-S2 support was merged with main last week. Enjoy.

closed 2021-02-10 by phoddie