Moddable-OpenSource/moddable #444: Listing on http://esp32.net/

wilberforce, 2020-09-07:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

**Describe**
A description of the feature you'd like to be added

**Why do you think this feature would be useful?**
Why you are requesting this feature? Who else might find it useful? If your request is related to a problem, please describe the problem as well.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered

phoddie, 2020-09-10:
You're right. We should. Do you have a contacts there or do we just reach out to them on Twitter?

wilberforce, 2020-10-07:
@phoddie 

Sorry - no contacts.

lprader, 2020-10-22:
I reached out to [@ESP32net](https://twitter.com/ESP32net) on Twitter about this last week. I'm still waiting to hear back, but I'll try again soon if I don't get a reply.

phoddie, 2020-12-30:
We did hear back from @ESP32net that an update is forthcoming. Fingers crossed.