Moddable-OpenSource/moddable #445: Documentation Issue: WebSocket Servers does not (and cannot) have a Write method.

LaurenceMason, 2020-09-09:
**Description**

WebSocket Servers do not have 'write' methods. Only the client do. The server invokes the client's write method to send a message to it. The documentation suggests that 'write' is a method of the server. Documentation excerpt: 

**write(message)**
The write function transmits a single WebSockets message. The message is either a String, which is sent as a text message, or an ArrayBuffer, which is sent as a binary message.

ws.write("hello");
ws.write(JSON.stringify({text: "hello"}));

The 'this' within the callback method is the client (not the server). For the server to be able to write to the connected clients outside the callback (eg. when some other event occurs), the clients need to be cached when the connection is made to the server. See the example below:

**const clients = [];**

const server = new Server({port:80});
server.callback = function (message, value) {
	switch (message) {
		case Server.connect:
			trace("main.js: socket connect.\n");
			trace(value);
			break;

		case Server.handshake:
			trace("main.js: websocket handshake success\n");
			**clients.push(this);**
			break;

		case Server.receive:
			trace(`main.js: websocket message received: ${value}\n`);
			this.write(value);		// echo
			break;

		case Server.disconnect:
			trace("main.js: websocket close\n");
			break;
	}
}

function sendMessage(s) {
	**clients.forEach(c => c.write(s));**
}

phoddie, 2020-09-09:
Good point. The documentation is incorrect there. I propose removing the `write` method from the server and revising [the server `callback` text](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#callbackmessage-value-1) as below.

That's better?

----

### `callback(message, value)`

The WebSocket server callback is the same as the WebSocket client callback with the addition of the "Socket connected" (`1` or `Server.connect`) message. The socket connected message for the server is invoked when the server accepts a new incoming connection. Messages cannot be sent until after the callback receives the WebSocket handshake complete message (`Server.handshake`).

The value of `this` is unique for each connection made to the server.  The `this` instance of the callback has the same `write` and `close` methods as the WebSocket Client. These methods are used to send data and to close the connection.

LaurenceMason, 2020-09-20:
Hello Peter,

Yes that is better. Sorry for the delay, if you were waiting for a response.

phoddie, 2020-09-21:
Excellent. Thank you for reviewing. The updated documentation will be part of a push this week.

closed 2020-09-21 by phoddie