Moddable-OpenSource/moddable #451: Valid switch + nested class syntax producing a SyntaxError exception

rwaldron, 2020-09-11:
**Description**

This is valid syntax: 
```
switch (0) { default: class x {} }
```

```
$ eshost -x 'switch (0) { default: class x {} }'
#### ChakraCore

#### engine262

#### JavaScriptCore

#### Moddable XS

SyntaxError: missing }

#### SpiderMonkey

#### V8

```

(Empty output is correct)

**Expected behavior**

No SyntaxError exception is thrown

phoddie, 2020-09-11:
I believe this is a duplicate of (or very similar to) Issue #412 (Switch statement contains class declaration is not parsed). That is fixed (and your example did not generate a Syntax Error when I tried it).

It may be that we haven't updated the `xst` binary to the most recent sources. If you [build `xst` from sources](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/xst.md#build), it  should work.

rwaldron, 2020-09-11:
Ah, that's very likely. I will close this!

closed 2020-09-11 by rwaldron

phoddie, 2020-09-11:
(Thank you for the report!)