Moddable-OpenSource/moddable #457: Reading REMOTE_IP on UDP socket crashes xsbug (and possibly the whole program)

JanVanBraeckel, 2020-09-19:
**Build environment:** Windows
**Target device:** NodeMCU ESP8266

**Description**
Creating a UDP socket and reading the "REMOTE_IP" of an incoming connection crashes xsbug. It possibly also crashes the whole program, but since the debugger crashes I can't strep through to check it.

Console output:
```
Exception (28):
epc1=0x40252b4b epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
NMAKE : fatal error U1077: '%MODDABLE%\build\bin\win\release\serial2xsbug.EXE' : return code '0xc0000005'
Stop.
```

main.js:
```
import { Socket } from "socket";

new Socket({
  port: 8266,
  kind: "UDP",
}).callback = function () {
  trace(this.get("REMOTE_IP"));
};
```

manifest.json:
```
{
  "include": [
    "$(MODDABLE)/examples/manifest_base.json",
    "$(MODDABLE)/examples/manifest_net.json"
  ],
  "modules": {
    "*": [
      "./main",
      "$(MODULES)/network/http/*",
    ],
  },
  "preload": [
    "http",
  ],
}
```

Python test script:
```
import socket

def main():
  remote_address = ('%addr%', 8266)
  message = 'test'
  sock2 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
  sock2.sendto(message.encode(), remote_address)
  sock2.settimeout(10)
  

if __name__ == '__main__':
  main()
```

**Steps to Reproduce**
1. Set up a program with a UDP socket where the callback tries to read the "REMOTE_IP"
2. Try to open the socket via an external program, for example a python script
3. See error

**Expected behavior**
I can read the REMOTE_IP and store it in a variable so I can include it in the Socket.write() method

phoddie, 2020-09-19:
The way to retrieve the IP address of the remote is different for TCP and UDP sockets.

For a TCP socket, the remote IP address never changes during the lifetime of the connection It is available with `this.get("REMOTE_IP")`.

For a UDP sockets, the remote IP address may be different for each packet received. Therefore, remote IP address is passed as an argument to the callback. The callback arguments for a UDP packet are:

```js
callback(message, value, address, port)
```

The remote IP address is in `address` and the sending port is in `port`. You can see an [example](https://github.com/Moddable-OpenSource/moddable/blob/54d8e94710ed6f8c310f046c4fbc83c967ccd346/modules/network/mdns/mdns.js#L312
) of this used in the mDNS implementation.

That said, calling `this.get("REMOTE"_IP")` on a TCP socket shouldn't crash. I'll fix that to return `undefined` in that case. It also looks like the documentation for `Socket` doesn't document most of this, so I'll update that as well.

JanVanBraeckel, 2020-09-20:
Oh I see!

I indeed failed to check if there were any extra arguments passed to the callback, since the documentation didn't mention any difference for the callback of a UDP socket.

I was searching and found this little snippet, so I thought I had found what I needed: https://github.com/Moddable-OpenSource/moddable/blob/5639abb24b6d725554969dc0be5822edb54a4a08/modules/network/socket/lwip/modSocket.c#L442

Thank you so much for the quick response, I love the work you are all doing on this project and hope to be able to contribute in the near future ☺

closed 2020-09-20 by JanVanBraeckel