Moddable-OpenSource/moddable #459: Windows build fails when building release tools but not debug

avivsiegel, 2020-09-23:
Windows 10, VS 2019 Native Tools Command Prompt; all patched and up-to-date

As I try to build the toolchain, I get a persistent issue with completing the build - it fails on building xsid.exe release (though the debug works well), and from there - things go south.
Git cloned using native Windows client, and it is not permission issue on the folder as other tools were built as expected (xsc.exe, xsl.exe) but then not much else: (I removed some lines from output and annotated with ...)

```
C:\zap\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f C:\zap\moddable\xs\makefiles\win\xsc.mak /s

... all is good too with debug builds for xsid, xsl, serial2xsbug, simulator, tools ...

C:\zap\moddable\build\makefiles\win>call C:\zap\moddable\build\bin\win\debug\mcconfig -d -m -p x-win C:\zap\moddable\tools\xsbug\manifest.json
C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f C:\zap\moddable\xs\makefiles\win\xsc.mak /s

... and it starts to fail in next one - and from there ony xsl, serial2xsbug and simulator are created and since xsid is not built others are failing ...

C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f C:\zap\moddable\xs\makefiles\win\xsid.mak /s
LINK : fatal error LNK1104: cannot open file 'C:\zap\moddable\build\bin\win\release\xsid.exe'
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX86\x86\link.EXE"' : return code '0x450'
Stop.

C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f C:\zap\moddable\xs\makefiles\win\xsl.mak /s
C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f serial2xsbug.mak /s
C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f simulator.mak /s
C:\zap\moddable\build\makefiles\win>nmake GOAL=release BUILD_DIR=C:\zap\moddable\build XS_DIR=C:\zap\moddable\xs /c /f tools.mak /s
# xsid commodettoBitmap.xsi
'C:\zap\moddable\build\bin\win\release\xsid' is not recognized as an internal or external command,
operable program or batch file.
NMAKE :  U1077: 'C:\zap\moddable\build\bin\win\release\xsid' : return code '0x1'
Stop.
C:\zap\moddable\build\makefiles\win>call C:\zap\moddable\build\bin\win\release\mcconfig -m -p x-win C:\zap\moddable\tools\xsbug\manifest.json
'C:\zap\moddable\build\bin\win\release\mcconfig' is not recognized as an internal or external command,
operable program or batch file.
```

I tried a fresh rebuilt in another folder, and refreshing the code base, as well as grab the code from github zip but to no avail.
Any idea?

andycarle, 2020-09-23:
Hi @avivsiegel, thank you for the detailed report. Let's see if we can get you back up and running.

This is a new one to me; I don't believe we've seen any reports of linker errors of this type on Windows. And I have verified that everything is working as expected on my machines. 

My understanding is that error LNK1104 from LINK indicates that the target output file has been locked. That can sometimes happen if the build tools hang on one attempt to build and then get wedged in a zombie state. Some folks on StackOverflow have reported that just rebooting has gotten them out of that state (unsatisfying as that answer is). But I would think that your idea of trying the build in a different folder would eliminate that possibility...

Microsoft [suggests that this error can be caused by antivirus software](https://docs.microsoft.com/en-us/cpp/error-messages/tool-errors/linker-tools-error-lnk1104?view=vs-2019) locking the newly created exe before it is finished being built. I have no idea why an antivirus package would choose to pick on xsid.exe in particular, but that's an idea that is at least worth exploring on your system.

You could also try turning on verbose output for the linker (fair warning: the output is **really** verbose). You can do that by editing [%MODDABLE%\xs\makefiles\win\xsid.mak](https://github.com/Moddable-OpenSource/moddable/blob/public/xs/makefiles/win/xsid.mak) at line 70 to include the `/VERBOSE` option:
`LINK_OPTIONS = /incremental:no /machine:I386 /nologo /subsystem:console /VERBOSE`

Perhaps the verbose output will give some hint?

Let me know if any of the above helps. If not, we'll keep digging.

&nbsp;- Andy

avivsiegel, 2020-09-24:
Andy - you are correct!

It was my antivirus - Bitdefender - so a reboot and disabling file safety monitoring did the trick.

Thank you for the quick response and resolution.

Aviv.

closed 2020-09-24 by avivsiegel

andycarle, 2020-09-24:
Great! I'm glad to hear it was something relatively easy.

I'll add a note about antivirus software and that error into the Getting Started guide.

&nbsp;- Andy