Moddable-OpenSource/moddable #465: commodetto/mini-drag doesn't work well.

basuke, 2020-09-26:
**Build environment:** macOS
**Target device:** Moddable Two

**Description**
I tried to build `commodetto/mini-drag` example and completed with no error. But I couldn't drag the button in the screen.

**Steps to Reproduce**
> mcconfig -d -m -p esp32/moddable_two

**Expected behavior**
I can drag `Drag Me` button by dragging.

**Other information**
When I set a breakpoint on the loop inside timer to observe the `point` value, it changes from 0 to 2 directly so that there's no chance to get click start state.

basuke, 2020-09-26:
FYI `pui/drag` works well on the same device.

phoddie, 2020-09-27:
Thanks for the report.

If i remember correctly, `mini-drag` was originally written for Moddable Zero, which has a resistive touch screen and therefore uses a different touch sensor than Moddable Two (which has a capacitive touch screen). The drivers generate slightly different events. The Piu code in the host works for both. The following change to the touch timer works in `mini-drag` on Moddable One and Moddable Two.

```js
Timer.repeat(() => {
	const points = touch.points;
	touch.read(points);
	const point = points[0];
	switch (point.state) {
		case 0:
		case 3:
			if (point.down) {
				delete point.down;
				point.target?.onTouchEnded(point.x, point.y);
				delete point.x;
				delete point.y;
				delete point.target;
			}
			break;
		case 1:
		case 2:
			if (!point.down) {
				point.down = true;
				point.target = dragger.contains(point.x, point.y) ? dragger : null;
				point.target?.onTouchBegan(point.x, point.y);
			}
			else
				point.target?.onTouchMoved(point.x, point.y);
			break;
	}
}, 17);
```

The change above also fixes a bug in the original example: if you start a drag outside the button and then drag through the button it would "picks up" the button. The optional-chaining operator works nicely as part of that a change.

We'll push this  in the coming days.

basuke, 2020-09-28:
Thanks. I'm working from primitive example, so that it takes a time to work with pui, but chances are to check the old examples with latest devices :) Should I close this bug right now or when I see the fix coming to my working copy?

phoddie, 2020-09-28:
Checking examples is great. There are so many that we cannot possibly check every example on every device all the time. Thank you for taking the time to report the issue.

Your call on when to close this bug. Now is fine with me.

(It is [Piu](https://gcide.gnu.org.ua/?q=Più&db=gcide&define=1), not Pui.)

basuke, 2020-09-28:
My bad. Say the word ten time: ```Array.from(new Array(10).keys()).forEach(_ => say('Piu'))```

closed 2020-09-28 by basuke