Moddable-OpenSource/moddable #467: Stack overflow (-1) on new Uint8Array in ssl_session.js?

avivsiegel, 2020-09-27:
My build env is Win10, VS19
Target is esp32/moddable_two
I noticed application stuck at new Request for SSL connection; all works well on Windows target, so I know call parameters are OK, as well as certs, protocolVersion, etc
When debugging and stepping through I noticed the exception is thrown in modules\crypt\ssl\ssl_session.js, constructor of SSLSession line 65 - 
		this.packetBuffer = new Uint8Array;
It is very consistent on this line.

Again, same code on win target works well (including initiation the connection and running the TLS calls), but not esp32.

Any idea?

phoddie, 2020-09-28:
The JavaScript stack is not the same size on Windows and ESP32. A good first step would be increasing the stack size on ESP32 to see if that helps. You can do that in your project manifest:

```json
	"creation": {
		"stack": 256
	}
```

The stack property is in units of slots, not bytes. On ESP32, slots are 16 bytes, so this is a  4 KB stack. Try making it 6 KB (384 slots)? (You can find documentation on the manifest configuration [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#creation).)

Where your application constructs the TLS session impacts the available stack size. If you don't mind, please post a screen grab of your stack from xsbug when it breaks on the stack overflow. Generally the default stack size is adequate, so I'm curious to understand why it is running out of stack here. Thanks.

avivsiegel, 2020-09-28:
Thank you for the quick response and support.
Well, this helped but not enough - when I extended the stack I ran out of memory, and your response to another issue led to extending static allocation, but then I ran into out of keys. Extending keys was the last thing needed for resolution and now my app works, with the following manifest creation parameters -

```
	"creation": {
		"static": 65536,
		"stack": 384,
		"keys": {
			"available": 64,
			"name": 53,
			"symbol": 3,
		},
	},

```
Watching the instrumentation, my app is using 41 keys, stack is going once in a while above 3600 so extension was needed.
Chunks is 15440 / 18176, and slots is hovering ~12000-14144 out of 14336 (looks tight here! so may need to extend).

[image]

Thanks!

phoddie, 2020-09-28:
Great - happy to hear you figured it out.

The default configuration values are deliberately small, so it is expected that some projects will need to adjust them.

closed 2020-09-28 by phoddie