Moddable-OpenSource/moddable #468: Support environment variable resolution in `config` section of manifest.json

basuke, 2020-09-28:
**Describe**
To hide private information, environment variable is a good solution. If we can use environment variable in `config` section of `manifest.json`, we can safely store ssid's password configuration into `manifest.json` and no need to pass it in command line argument every time.

```
"config": {
  "ssid": "$(MY_HOME_SSID)",
  "password": "$(MY_SECRET_WIFI_PASSWORD)"
}
```

**Why do you think this feature would be useful?**
What configuration is required for the project is very important information to describe the structure of it. Those should be listed up in `manifest.json`. Command line arguments are handy, but it should be kind of optional, for instance to override those in `manifest.json` for debugging purpose.

Also it is not good idea to put secure information in command line because they are also stored in the history of shell.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered

phoddie, 2020-09-28:
You raise some interesting point. Security is tricky (for me, at least), and often in conflict with convenience, so I'd like to explore this a bit.

Our intention with command line configuration parameters is precisely what you describe - that they are primarily a convenience for development. The manifest is the preferred location for most configuration static data.

We assume that the host (aka application, the thing you flash onto the device) is responsible for the Wi-Fi configuration. You would not typically create a separate build of the host for each lightbulb manufactured. You would configure the software with your Wi-Fi access point and password. Those would be stored in the preferences on the device as part of some provisioning step. A well written host would obfuscate or encrypt that information.

Using an environment variable solves some problems, but not all:

- the firmware image is still bound to a single device (which is OK for personal projects)
- the SSID and password are still stored in the clear as part of the firmware image
- I imagine an attack where a manifest puts many unnecessary environment variables into the configuration of a firmware image. They might do that in an example. When you build that, the installed software gets access to the those environment variables from your computer. An attacker might be able to use that information to compromise your system.

Taking a step back.... implementing what you propose is certainly possible. There are some details about escaping that would  need to be resolved (how can you put the `$()` sequence into a config value. But, I'm not sure we should do this.

basuke, 2020-10-02:
I understand what you care about security and you are right. This enhancement is only for development convenience, as command line config does. This won't solve and shouldn't be used to solve the problems you listed above. On the other hand, I think this doesn't lower the security level of build because it is fundamentally same with passing those values from command line argument.

I'm convincing great productivity improvement with this. To be honest, another idea was already in my mind, which is support of .env file in the project root. It is very popular workflow for people from other environment. If file is there, the file is loaded and those definitions are used as if environment variables are defined. Adding this file in .gitignore, it's easier to make the repository to be shared among others.

Thinking about the security, if `mcconfig` support environment variable, it should be disabled by command line option. On final build for the serious production, it should be prohibited explicitly. Also logging out those use environment variables are some help for auditing.

phoddie, 2020-10-02:
From your original comments, I had understood this to be a security concern:

> To hide private information, environment variable is a good solution

> it is not good idea to put secure information in command line because they are also stored in the history of shell

Your more recent comments appear to have a different priority:

> This enhancement is only for development convenience

> I'm convincing great productivity improvement with this

I understand that it is common for some kinds of web development to store passwords and other private configuration data in environment variables. This is done to avoid storing them in files -- make files, manifests, source code -- that could be accidentally committed to a source code repository, which is particularly problematic when using a public repository such as GitHub.

Your first example above shows using environment variables in a manifest to avoid including a password in a file that is committed to Git.

```json
"config": {
  "ssid": "$(MY_HOME_SSID)",
  "password": "$(MY_SECRET_WIFI_PASSWORD)"
}
```

These values are taken from the current environment of the computer and transferred to the target device as part of the virtual machine image.

A project in the Moddable SDK is made up of many manifests  - the project manifest, the MCU manifest, the device target manifest, and manifests for many of the individual modules. Each manifest could potentially import an arbitrary number of environment variables. The [mDNS manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/mdns/manifest.json) could be modified to scrape passwords from the environment:

```json
"config": {
  "passwords": {
    "aws": "$(AWS_PASSWORD)",
    "azure": "$(AZURE_PASSWORD)",
    "gcloud": "$(GOOGLE_CLOUD_PASSWORD)",
  }
}
```

Any matches there will be delivered to the software running on the device. 

Of course, you can audit for that. But, in a system with a large number of modules mistakes happen. There are malicious actors. It would be more secure if this powerful capability were simply unavailable to build scripts. That's one fewer source of mistakes for developers to be concerned with.

As it is common practice to store passwords in environment variables, I do agree that making it possible to use those in the Moddable SDK build would be convenient. That can be done. A project that works with AWS can be built to include that:

```
export MY_SECRET_WIFI_PASSWORD=secret
mcconfig -d -m -p esp32/moddable_two password=$MY_SECRET_WIFI_PASSWORD
```

By being explicit, this avoids a malicious module scraping information from environment variables. It is more characters to include on the command line, but the shell provides many tools for developers to create shortcuts.

But, as I said above, this is  all kind of academic because it isn't what real products do. In a real product, this kind of configuration data is usually provisioned separately from the code in the firmware image. A common mechanism for that, especially on small devices, is preferences. A real product has ways to set the preferences -- whether as part of manufacturing or as part of user set-up. The Moddable SDK Preferences are convenient to use:

```js
Preference.set("domain", "name", "value");
let value = Preference.get("domain", "name");
```

But, writing the provisioning code is usually not the first thing a developer does because it takes time. It might mean building embedded web pages or a mobile app. What would help is a way to set the preference on the device from the computer using the command line. Imagine something like:

```
mcpref /dev/cu.SLAB_USBtoUART -set domain name value
```

That approach allows the developer to write code for the device that works exactly the same as the final product. It also avoids having any passwords in the firmware image itself. It would also be useful beyond passwords -- to configure various other product behaviors from the command line:

```
mcpref /dev/cu.SLAB_USBtoUART -set light color red
mcpref /dev/cu.SLAB_USBtoUART -set light brightness 50
```

wilberforce, 2020-10-02:
@phoddie although it defeats the purpose of the original question and option on mcpref to take a json file as arg to do the setup with one lot of flashing would be good.

phoddie, 2020-10-02:
@wilberforce - Yes, good point. The preference setting could be batched (with additional command line arguments, a file, etc). That would be faster, for sure.

basuke, 2020-10-02:
> As it is common practice to store passwords in environment variables, I do agree that making it possible to use those in the Moddable SDK build would be convenient. That can be done. A project that [...]

That's exactly what I did on my YouTube video :) https://youtu.be/I0SEBPTzQ3Q?t=386

> But, as I said above, this is all kind of academic because it isn't what real products do. In a real product, this kind of configuration data is usually provisioned separately from the code in the firmware image.

I have no objection about this, but is Moddable SDK only for professional use? I remember you mentioned that it's there to deliver casual way to people who want to modify things. This enhancement is nice to have feature for that purpose to be casual and friendly for rest of us, even if that is optional (off by default).

> But, writing the provisioning code is usually not the first thing a developer does because it takes time. It might mean building embedded web pages or a mobile app. What would help is a way to set [...]

Yeah, this is great idea. This solves some part of the issue. I still prefer having an option for environment variable, but if that's not the option, I don't have objection to close the issue.

basuke, 2020-10-03:
> As it is common practice to store passwords in environment variables, I do agree that making it possible to use those in the Moddable SDK build would be convenient. That can be done. A project that [...]

That's exactly what I did on my YouTube video :) https://youtu.be/I0SEBPTzQ3Q?t=386

> But, as I said above, this is all kind of academic because it isn't what real products do. In a real product, this kind of configuration data is usually provisioned separately from the code in the firmware image.

I have no objection about this. Security for real should be always the first priority. I didn't think about that so much because I'm hobbyist and no need to think about them.

Maybe we need to re-define `developer's convenience` because that's different. For me, as a Sunday hobbyist, it is like just typing `make` after moving to the directory. I don't touch my project every day, weekly or monthly. Command line option is a little bit hard to work with this cadence. That's the initial passion for this issue. Then I thought supporting `.env` is a good option, so I patched. But then I realized environment variables are only supported in SOME part of `manifest.json`. Then for me it was time to make an issue.

This enhancement is nice to have feature for that purpose, even if that is optional (off by default).

> But, writing the provisioning code is usually not the first thing a developer does because it takes time. It might mean building embedded web pages or a mobile app. What would help is a way to set [...]

Yeah, this is great idea. This solves some part of the issue. I still prefer having an option for environment variable.

phoddie, 2020-10-05:
The latest Moddable SDK sources contain an enhancement to `serial2xsbug` to set preferences from the command line. Eventually this needs to be wrapped in a nicer command line interface, but this gets the basic mechanism in place to experiment with and refine.

This sets two preferences, both in the domain "foo". The first is with the name "bar" and value "This is a test.". The second is with the name "barbar" and the value of the $ID_PATH environment variable. You can set several preferences with a single command line.

```
serial2xsbug /dev/cu.SLAB_USBtoUART 460800 8N1 -pref foo.bar="This is a test." -pref foo.barbar=$IDF_PATH
```

This also works with ESP8266 targets. The only difference is the baud rate:

```
serial2xsbug /dev/cu.SLAB_USBtoUART 921600 8N1 -pref foo.bar="This is a test." -pref foo.barbar=$IDF_PATH
```

Implementation details:
- all preference values are set as strings
- the device must be running a debug build
- xsbug must be running on the host

> Reminder: update $MODDABLE, rebuild tools, install debug example on the device, themn you can use `-pref`

phoddie, 2020-10-22:
i believe this topic has been addressed as much as practical at this point, so I'm closing this .

closed 2020-10-22 by phoddie