Moddable-OpenSource/moddable #470: Change the value of hostname

ederle, 2020-09-29:
Hi, how to change the value of hostname for ESP32 & 8266 ? The default value is espressif. Is there an API for that ?

Thanks

phoddie, 2020-09-29:
Hello. I'm not sure what you are referring to by "hostname". Would you be more precise?

FWIW - The Moddable SDK does not assign a hostname to the device. How are you observing that a the hostname is "espressif"?

domede, 2020-09-29:
>Hello. I'm not sure what you are referring to by "hostname". Would you be more precise? ...

@phoddie my esp32 service is connected to my dsl box and i

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ederle, 2020-09-30:
hi, thank you for your reply :-)

When i connect my esp32 device that run my moddable app to my dsl box, i see in the devices list connected the device name associated with ip address (DNS) for my eps32 device the name is "espressif". 
And when I do a "ping espressif", I have an answer from my device.

I searched for the word "espressif" in the ~ / esp32 / esp-idf folder and found in some C files: 
```c
netif-> hostname = "espressif";
```
But I did not find how to change the value of "hostname ".

Do you know if it's possible to change this value?

Thanks :-)

phoddie, 2020-09-30:
Thank you for the details.

I never tried to set that hostname on the Espressif devices. I did some investigation and found it can be done using `tcpip_adapter_set_hostname`. It is necessary to wait for the Wi-Fi interface to come up, which does not happen immediately after calling `esp_wifi_start`, but it succeeds after receiving the `SYSTEM_EVENT_STA_START` event.

I've made a small change to the Moddable SDK Wi-Fi code to support setting the hostname in the project manifest's defines section. Here's a trivial example;

	"defines": {
		"wifi": {
			"hostname": "#tada"
		}
	},

We'll push the change to this repository a little later today.

On a somewhat related note, the Moddable SDK implements mDNS so you can claim a hostname that way (and automatically resolve name conflicts). Here's a basic example:

https://github.com/Moddable-OpenSource/moddable/blob/e1fe8b4909e5a0518dc7b04e56caea1ab73ea23e/examples/network/mdns/httpserver/main.js#L31-L36

ederle, 2020-10-01:
Thank you very much. I test as soon as possible and I inform you of the result.
Have a nice day :-)

ederle, 2020-10-01:
Some weeks ago, i've tried this code (below) immediately after the wifi connection was ready. It seems working fine but my esp32 device had no hostname on my network.
 
```javascript
_claimHostname() {
    if( this._hostname.length>0 ) {
        const mdns = new MDNS({hostName: this._hostname}, (message, value) => {
            switch( message ) {
                case 1:
                    if( this.debug ) {
                        trace(`${Date()}: MDNS - claimed hostname is "${value}"\n`);
                    }
                    break;
                case 2:
                    if( this.debug ) {
                        trace(`${Date()}: MDNS - failed to claim "${value}", try next\n`);
                    }
                    break;
                default:
                    if( message<0 ) {
                        if( this.debug ) {
                            trace("${Date()}: MDNS - failed to claim, give up\n");
                        }
                    }
                    break;
            }
        });
    }
}
```
I've copied this piece of code from moddable documentation: network / class MDNS.

---
... and congratulations: this is the second personal project I have done with Moddable SDK and it works really well!

phoddie, 2020-10-01:
Glad to hear you are having good success with the Moddable SDK. If you decide to make any of your personal projects available, please let me know. It is always interesting to see how people are using the SDK.

Regarding mDNS, I adapted your code slightly and added it at the end of $MODDABLE/examples/network/http/httpget/main.js:

```js
import MDNS from "mdns"
const _hostname = "test";
const debug = true;

const mdns = new MDNS({hostName: _hostname}, (message, value) => {
	switch( message ) {
		case 1:
			if( debug ) {
				trace(`${Date()}: MDNS - claimed hostname is "${value}"\n`);
			}
			break;
		case 2:
			if( debug ) {
				trace(`${Date()}: MDNS - failed to claim "${value}", try next\n`);
			}
			break;
		default:
			if( message<0 ) {
				if( debug ) {
					trace("${Date()}: MDNS - failed to claim, give up\n");
				}
			}
			break;
	}
});
```

That worked as expected for me. The hostname with mDNS is always in the ".local" domain, so I did this to check it:

```
> ping test.local
PING test.local (10.0.1.16): 56 data bytes
64 bytes from 10.0.1.16: icmp_seq=0 ttl=128 time=327.895 ms
64 bytes from 10.0.1.16: icmp_seq=1 ttl=128 time=9.557 ms
```

Not all computers resolve ".local" addresses -- macOS definitely does. For Windows, I believe it requires additional configuration.

ederle, 2020-10-02:
It was my mistake, I did not add `.local` to the hostname in my pings. I will test again this weekend.
Thanks :-) Have a nice day .

ederle, 2020-10-09:
All works fine (macOS)  ! 
Thanks a lot :-)

closed 2020-10-09 by ederle