Moddable-OpenSource/moddable #471: Cannot set up on Catalina

thomasqbrady, 2020-10-01:
On step five of the environment setup for Mac instructions, and I get:

# xsc debug : cc xsc
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsre.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsc.o'
make[1]: *** [/Users/[username]/Development/moddable/build/bin/mac/debug/xsc] Error 1
make: *** [debug] Error 2

where [username] is actually my username. I have XCode and the developer tools installed.

andycarle, 2020-10-02:
Interesting; I don't think we've seen this particular failure before. Let's see if we can sort this out.

The error you're getting there would seem to suggest that clang cannot create those output files. After it fails there, does the directory `/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/` exist? Perhaps the `mkdir` failed first and now clang can't generate those .o files.

One other long shot: this error is very similar to [another issue we saw recently on Windows](https://github.com/Moddable-OpenSource/moddable/issues/459), where the user's Bitdefender antivirus software was blocking the creation of new executables. Any chance you have some antivirus program running that could be interfering with the build?

&nbsp;- Andy

thomasqbrady, 2020-10-03:
Yes, that directory does get created, but is empty. 

And yes, my employer does require antivirus software. I think we’re using Falcon and Crowdstrike.

I could compile on another machine and copy the executable over, right? Except that my other machine is on a Big Sur beta...

andycarle, 2020-10-05:
Thanks for checking back in on this issue. The similarity to other antivirus interference we've seen is certainly suspicious. 

I'll try to make some time today or tomorrow to install that antivirus package on my Catalina machine as a test. If I can verify that that's the issue, perhaps we can find a way to work around the conflict.

And yes, you could likely build the Moddable SDK tools on another machine and copy them over. But I suspect you'll just keep running into this problem as you try to build apps using those tools...

thomasqbrady, 2020-10-06:
So strange. I mean, antivirus applications are awful, for sure, but I've never had any trouble building anything else on this particular setup. I'm happy to do anything I can to test/confirm/debug/log, by the way.

andycarle, 2020-10-20:
Hi @thomasqbrady: a quick update on this on my side.

I was finally able to get a trial version of Crowdstrike Falcon up and running. But I have not been able to reproduce any build issues with the AV software monitoring. And it does not seem to be generating any warning or "detections" in its admin console. The antivirus idea may be a red herring. 

If you're willing to work with me on this a bit, let's run through some additional possibilities. 

Could you:
1. Pull updated sources.
2. Delete the $MODDABLE/build/bin and $MODDABLE/build/tmp directories.
3. `export VERBOSE=1` in your shell.
4. Retry building the Moddable SDK tools.

Perhaps there will be some other clue in the verbose output when that fails. If you could share the build output here (or email it to me if there's any sensitive info in that you want to avoid posting publicly; I'm andy@moddable.com), I'd love to take a look and see if I can sort it out.

Cheers!
&nbsp;- Andy

thomasqbrady, 2020-10-20:
No problem! I appreciate the help, and I'm glad to do what I can.

Following the instructions above, I got:

`➜  mac git:(public) make
make -f ~/Development/moddable/xs/makefiles/mac/xsc.mk
mkdir -p ~/Development/moddable/build/tmp/mac/debug/xsc
mkdir -p ~/Development/moddable/build/bin/mac/debug
# xsc debug : cc xsc
cc -mmacosx-version-min=10.7  -framework CoreServices ~/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsre.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsc.o -o /Users/[username]/Development/moddable/build/bin/mac/debug/xsc
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsre.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsc.o'
make[1]: *** [/Users/[username]/Development/moddable/build/bin/mac/debug/xsc] Error 1
make: *** [debug] Error 2`

andycarle, 2020-10-21:
Interesting... this is failing at a slightly different place than I had thought, based on your first message. This is the linker trying to assemble the xsc executable from the listed object files, presumably after having successfully built them.

Okay, next things to check:
1) If you look in the directory `~/Development/moddable/build/tmp/mac/debug/xsc/` do you see the referenced .o files? xsBigInt.o, xsCommon.o, etc.? 
2) Just to be 100% certain: `/Users/[username]/Development/moddable` and `~/Development/moddable` are the same directories, yeah? :)
3) Could you check the permissions on the `~/Development/moddable/build/tmp/mac/debug/xsc/` directory and its contents using `ls -l`? The directory should be owned by the user you're using to do the build and have 755 permissions (`rwxr-xr-x`) and the object files should be owned by the user you're using to do the build and have 644 permissions (`rw-r--r--`)
4) Your build output here is for a partially complete build (it has already build those .o files.) Could you do a full clean by deleting the $MODDABLE/build/bin and $MODDABLE/build/tmp directories, and then send me the full build output with `export VERBOSE=1` set? I just want to make sure there aren't any warnings or errors when generating those object files.
5) Out of curiosity, does your actual account name (the `[username]` bit redacted in the above) contain any non-alphanumeric or non-ASCII characters? Anything like a space or symbol or a unicode character?

Thanks! We'll get to the bottom of this.
&nbsp;- Andy

thomasqbrady, 2020-10-22:
1. No, xsc exists, but is empty
2. Yes. : D
3. drwxr-xr-x
4. That's strange... I deleted those directories like you asked in step 2 in your previous comment. Output below.
6. No, just alphanumeric. It's a corporate username that they're not keen on us exposing.

`
➜  mac git:(public) make
make -f ~/Development/moddable/xs/makefiles/mac/xsc.mk
mkdir -p ~/Development/moddable/build/tmp/mac/debug/xsc
mkdir -p ~/Development/moddable/build/bin/Mac/debug`
`# xsc debug : cc xsc`
`cc -mmacosx-version-min=10.7  -framework CoreServices ~/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsre.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsc.o -o /Users/b161861/Development/moddable/build/bin/mac/debug/xsc
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsre.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o'
clang: error: no such file or directory: '/Users/b161861/Development/moddable/build/tmp/mac/debug/xsc/xsc.o'
make[1]: *** [/Users/b161861/Development/moddable/build/bin/mac/debug/xsc] Error 1
make: *** [debug] Error 2`

andycarle, 2020-10-23:
Now *that* is weird.

If that is the immediate output of running `make` right after deleting the `bin` and `tmp` directories, something very strange is happening. The output there is showing an attempt at linking together the final executable happening without any of the object files having been built.  That step should be dependent upon the existence of the object files.

Two ideas:

1) What version of `make` are you using? When I run `make --version` I get:

```
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
```

2) I've put together a version of the xsc makefile that includes some additional logging to try to sort this out. [It is attached here.](https://github.com/Moddable-OpenSource/moddable/files/5426900/xsc.mk.zip)

Please try:
 - unzipping that
 - replacing `$MODDABLE/xs/makefiles/mac/xsc.mk` with the new version of `xsc.mk`
 - once again deleting `$MODDABLE/build/bin` and `$MODDABLE/build/tmp`
 - `cd $MODDABLE/build/makefiles/mac`
 - `make`

And then send me the output of that. (Sorry, you'll have to do some redacting on the paths I'm printing there.) I'm hoping that will give some insight into why those dependencies are not being built.

thomasqbrady, 2020-10-23:
1. Same output as yours
2. On quick inspection, I don't see a difference, but:

`make -f ~/Development/moddable/xs/makefiles/mac/xsc.mk
TMP_DIR: ***~/Development/moddable/build/tmp/mac/debug/xsc***
OBJECTS: ***~/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsre.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsc.o***
In link stage: ***~/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsre.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o ~/Development/moddable/build/tmp/mac/debug/xsc/xsc.o***`
`# xsc debug : cc xsc`
`clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsBigInt.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCode.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsCommon.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsdtoa.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsLexical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsre.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScope.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsScript.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSourceMap.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsSyntaxical.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsTree.o'
clang: error: no such file or directory: '/Users/[username]/Development/moddable/build/tmp/mac/debug/xsc/xsc.o'
make[1]: *** [/Users/[username]/Development/moddable/build/bin/mac/debug/xsc] Error 1
make: *** [debug] Error 2`

thomasqbrady, 2020-10-23:
Figured it out! I had my MODDABLE environment variable set to: `export MODDABLE="~/Development/moddable"`

Instead, setting it to: `export MODDABLE="/Users/[username]/Development/moddable"` works just fine.
🤦

phoddie, 2020-10-23:
@thomasqbrady - Thanks for persevering.

@andycarle - Is there a way we can detect this problem, to help out others who might do something similar?

andycarle, 2020-10-23:
@thomasqbrady Aha! That's great that you sorted it out!

I had an inkling that the issue was related to the ~ in your paths, so early in this process I tested `export MODDABLE=~/Development/moddable`. That version *does* work just fine; but, yes, putting quotes around it will cause the ~ to not be properly evaluated.

As an aside: this probably means that you have a stray directory somewhere (probably in `$MODDABLE/build/makefiles/mac`) called, literally, `~/Development/moddable/build/tmp/mac/debug/` that has those object files we were looking for.

@phoddie Yes, I think I can sort something out to throw an error on this condition. (We can also clarify in the docs that a full absolute path should be used.)

andycarle, 2020-10-24:
I've pushed a change to provide an earlier and clearer error message for this case and for similar errors with the `MODDABLE` environment variable. It will be present in our next open source push.

Thank you again for sticking it out through this, @thomasqbrady! I'm going to go ahead and close this issue.

closed 2020-10-24 by andycarle