Moddable-OpenSource/moddable #472: BLE build issues on windows

wilberforce, 2020-10-03:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

> You may use this template to file a report of a security vulnerability. If you prefer to report the issue privately, please send the report by email to info@moddable.com.

**Build environment:** macOS, Windows, or Linux
**Target device:** Moddable One, Moddable Two, Moddable Three, desktop simulator, NodeMCU ESP8266, NodeMCU ESP32, etc.

**Description**
I have a test harness that allows  a BLE stub to build on windows so the same code base can be used ESP32/windows

**Steps to Reproduce**
```
F:\Projects\moddable\modules\network\ble\modBLE.h(103): error C2059: syntax error: ';'
F:\Projects\moddable\modules\network\ble\modBLE.h(104): error C2059: syntax error: '}'
F:\Projects\moddable\modules\network\ble\modBLE.h(124): error C2061: syntax error: identifier 'modMessageDeliver'
F:\Projects\moddable\modules\network\ble\modBLE.h(127): error C2059: syntax error: '}'
F:\Projects\moddable\modules\network\ble\modBLE.h(134): error C2059: syntax error: ';'
F:\Projects\moddable\modules\network\ble\modBLE.h(135): error C2059: syntax error: '}'
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2146: syntax error: missing ')' before identifier 'callback'
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2081: 'modMessageDeliver': name in formal parameter list illegal
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2061: syntax error: identifier 'callback'
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2059: syntax error: ';'
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2059: syntax error: ','
F:\Projects\moddable\modules\network\ble\modBLE.h(139): error C2059: syntax error: ')'
F:\Projects\moddable\modules\network\ble\btutils.c(106): error C2065: 'modCriticalSectionDeclare': undeclared identifier
F:\Projects\moddable\modules\network\ble\btutils.c(107): warning C4013: 'modCriticalSectionBegin' undefined; assuming extern returning int
F:\Projects\moddable\modules\network\ble\btutils.c(108): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(109): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(109): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(112): warning C4013: 'modCriticalSectionEnd' undefined; assuming extern returning int
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2146: syntax error: missing ')' before identifier 'callback'
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2081: 'modMessageDeliver': name in formal parameter list illegal
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2061: syntax error: identifier 'callback'
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2059: syntax error: ';'
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2059: syntax error: ','
F:\Projects\moddable\modules\network\ble\btutils.c(115): error C2059: syntax error: ')'
F:\Projects\moddable\modules\network\ble\btutils.c(126): error C2065: 'modCriticalSectionDeclare': undeclared identifier
F:\Projects\moddable\modules\network\ble\btutils.c(128): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(129): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(130): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(131): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(132): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(133): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(137): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(141): warning C4013: 'modMessagePostToMachine' undefined; assuming extern returning int
F:\Projects\moddable\modules\network\ble\btutils.c(141): error C2037: left of 'the' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(141): error C2037: left of 'callback' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(141): error C2037: left of 'refcon' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(147): error C2065: 'modCriticalSectionDeclare': undeclared identifier
F:\Projects\moddable\modules\network\ble\btutils.c(149): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(150): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(151): error C2037: left of 'entries' specifies undefined struct/union 'modBLEMessageQueueRecord'
F:\Projects\moddable\modules\network\ble\btutils.c(151): error C2037: left of 'next' specifies undefined struct/union 'modBLEMessageQueueEntryRecord'
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.21.27702\bin\HostX86\x86\cl.EXE"' : return code '0x2'
```

**Expected behavior**
Compile without errors.

Appears to be related to this commit:

https://github.com/Moddable-OpenSource/moddable/commit/73bc7df45db69c1fc04db27ceeea9a95da239005

phoddie, 2020-10-03:
We'll take a look. But... FYI... the BLE code is not intended to run on Windows, so this isn't something we would necessarily expect to work.

bfriedkin, 2020-10-03:
@wilberforce -

Sorry for the hiccup. We will get a fix into the next push. In the meantime, you can build with the attached patch.

Regards,
Brian

[ble.txt](https://github.com/Moddable-OpenSource/moddable/files/5322400/ble.txt)

wilberforce, 2020-10-04:
> [ble.txt](https://github.com/Moddable-OpenSource/moddable/files/5322400/ble.txt)

Thanks. That works!

wilberforce, 2020-10-06:
@phoddie @mkellner 
Thanks - confirmed builds ok for windows sim.

closed 2020-10-06 by wilberforce