Moddable-OpenSource/moddable #473: display issues on new m5-Stick C hardware

wilberforce, 2020-10-07:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report, not to follow the template.

> You may use this template to file a report of a security vulnerability. If you prefer to report the issue privately, please send the report by email to info@moddable.com.

**Build environment:** Windows
**Target device:** new m5-Stick C

**Description**
Getting display smudging on the new hardware. Works ok on the older hardware. Photo to follow

**Steps to Reproduce**
Interestingly the sample: https://github.com/Moddable-OpenSource/moddable/tree/public/examples/drivers/m5stickc-imu
works fine.

The app I'm building uses both BLU and WIFI.

This seems to be related to changes in the last 3 weeks - I've not been able to narrow it down yet.

**Expected behavior**
Both old and new hardware to display cleanly.

**Images**

phoddie, 2020-10-07:
@wilberforce, that's an artifact I'm not sure I could generate in software (at least not easily!). I'm going to guess this is a hardware configuration problem. (I don't have an M5 Stick, so I can only guess)

The recent addition of support for M5Stack CORE2 included changes to the driver code for the AXP192 power management chip. I believe that chip controls power to the display. The AXP192 code was reorganized to put the device specific code into [setup-target.js](https://github.com/Moddable-OpenSource/moddable/blob/ece6a84f3190d1191bf4698c656aad0755d6d5e3/build/devices/esp32/targets/m5stick_c/setup-target.js#L122-L163). Maybe you can review those changes?

It is curious that you don't see a problem with the IMU ball example. That might be because it has a 25% gray background color, which is much less bright than your project's white background. 

@meganetaaan seems to understand the AXP192. Maybe he can help?

wilberforce, 2020-10-08:
@phoddie 
> The recent addition of support for M5Stack CORE2 included changes to the driver code for the AXP192 power management chip
Thanks for the direction - I'll roll this back and see if this makes a difference.

>It is curious that you don't see a problem with the IMU ball example. That might be because it has a 25% gray background color
Another area to try - the problem still exists with the brightness so between 7-15. I'll try the actual background colour too.

Thanks for the hints - exactly was I was after!

wilberforce, 2020-10-09:
@phoddie @meganetaaan 

I've got to the bottom of it - it's when I call the `brightness` setter.

I'm passing the value 10, so it's writing to 0x28 with the value 160.

````
brightness=10;
global.power.brightness = brightness;
```

https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stick_c/setup-target.js#L139-L142

As soon as this happens, the display smudges...

phoddie, 2020-10-09:
Cool! 

FWIW - Tthe range of brightness should be 0 to 100. That's what the backlight control on Moddable Two uses:

https://github.com/Moddable-OpenSource/moddable/blob/ece6a84f3190d1191bf4698c656aad0755d6d5e3/build/devices/esp32/targets/moddable_two/setup-target.js#L12-L19

So the 10 you are passing is valid. I guess the question that needs to be sorted out is the range of the power register at 0x28.

wilberforce, 2020-10-11:
Existing:
```
set brightness(brightness) {
    const b = (brightness & 0x0f) << 4;
    this.writeByte(0x28, b);
  }
```

https://github.com/m5stack/M5StickC/blob/master/src/AXP192.cpp#L155-L163

```
uint8_t buf = Read8bit( 0x28 );
    Write1Byte( 0x28 , ((buf & 0x0f) | (brightness << 4)) );
```

phoddie, 2020-10-12:
Thanks for the reference to the M5StickC code. The complete function is helpful:

https://github.com/m5stack/M5StickC/blob/a5a5ca9743a8ff0176072aacba73e9a5fc701d85/src/AXP192.cpp#L155-L163

As you noted above, the low nibble of the register is getting overwritten. But, the range is 0 to 12 is important too. It is 0 to 12, so we should map the 0 to 100 input. Something like this?

```js
set brightness(value) {
	if (value <= 0)
		value = 0;
	else if (value >= 100)
		value = 12;
	else
		value = (value / 100) * 12;
	this.writeByte(0x28, (this.readByte(0x28) & 0x0F) | (value << 4));
}
```

meganetaaan, 2020-10-12:
Sorry to be late! I didn't know that so many has been updated on the original repos since my first PR for adding M5StickC target.

* The change of brightness setter is one of them. But I wonder why current original sets 12 as upper limit. I remember valid register value is between 7 and 15 when I made my commit. Once you try setting 0-6 you will know it leads the same as dark as 7. Below worked for me.
```js
	set brightness(value) {
		if (value <= 0)
			value = 7;
		else if (value >= 100)
			value = 15;
		else
			value = (value / 100) * 8 + 7;
		this.writeByte(0x28, (this.readByte(0x28) & 0x0F) | (value << 4));
	}
```

* FWIW I've noticed the [initialization sequence](https://github.com/m5stack/M5StickC/blob/a5a5ca9743a8ff0176072aacba73e9a5fc701d85/src/AXP192.cpp#L8) also has been changed a lot. Current code could have another issue about compatibility of old and new hardware. But unfortunately I cannot confirm them because I don't have new one :(

wilberforce, 2020-10-13:
@meganetaaan 

That change works fine. I'll do a pull request

closed 2020-10-30 by phoddie