Moddable-OpenSource/moddable #475: Build error in "examples/piu/bars"

stc1988, 2020-10-15:
**Version** October 14, 2020
**Build environment:** Windows
**Target device:** M5stack core2

**Description**
Build error in "examples/piu/bars"

**Steps to Reproduce**
1. build "examples/piu/bars"
build command is `mcconfig -d -m -p esp32/m5stack_core2`

2.  build errror `moddable\examples\piu\bars\main.js(47): error: missing expression!`

**Other information**
https://github.com/Moddable-OpenSource/moddable/blob/6165841a7725bc6488528d43704ca822204451b1/examples/piu/bars/main.js#L45-L47

I delete trailing commas, build is success

phoddie, 2020-10-15:
Good find. Thank you for reporting the fix! We'll get that into our next SDK update.

This error was previously undetected by our XS JavaScript engine. It was fixed just yesterday. See issue #315, "XS allows trailing comma in Expressions," for details.

stc1988, 2020-10-21:
I confirmed fixed in October 20, 2020.

closed 2020-10-21 by stc1988