Moddable-OpenSource/moddable #478: mcconfig alert MODDABLE: variable not found

WebSamuraiDaisuke, 2020-10-20:
I have followed step 8 of Geting Started for Windows. I have tried `cd %MODDABLE%\examples\helloworld` and `mcconfig -d -m -p win`, the console shows me `MODDABLE: Variable not found!`. My goal is to install the example to M5stack. How can I solve this issue?

andycarle, 2020-10-20:
It sounds like something must have gone wrong with step 4, where you need to set the MODDABLE environment variable. There are two likely possibilities:

1) You set the environment variable in a single Command Prompt instance rather than in System Properties and then changed to a different Command Prompt instance. To properly set the environment variables, please follow the steps in [this guide](https://www.architectryan.com/2018/08/31/how-to-change-environment-variables-on-windows-10/).

2) You may not have started a new Command Prompt instance after setting up the environment variables in System Properties. Newly set global/user environment variables won't take effect until the next time you start Command Prompt.

Please try those steps and let me know how it goes!
&nbsp;- Andy

WebSamuraiDaisuke, 2020-10-21:
@andycarle Thanks for your reply.

But the next issue has come. mcconfig says `manifest not found`.
I have executed `mcconfig C:\Users\my_name\Projects\moddable\examples\helloworld

[image]

andycarle, 2020-10-21:
@WebSamuraiDaisuke Happy to help; glad that we're making progress here.

I think the value of your `MODDABLE` environment variable may be incorrect still. Can you please verify that it contains the exact path to your Moddable SDK directory? When setting the variable, I recommend using the "Browse Directory" button to ensure the exact right path is chosen:

[image]

After double-checking that and opening a new Command Prompt, you can verify that the setting is correct by typing `echo %MODDABLE%` at the Command Prompt. Could you paste the exact output of that command here (or email me at andy@moddable.com if you don't want to post sensitive information here)?

Thank you!
&nbsp;- Andy

andycarle, 2020-10-21:
@WebSamuraiDaisuke One other suggestion. Could you please try:
1) Deleting the `%MODDABLE%\build\bin` and `%MODDABLE%\build\tmp` directories.
2) Rebuilding the Moddable SDK tools.
3) Retrying the `helloworld` build.

It is possible that you have a partial build in progress still using your old environment variable settings. Clearing the old build will make sure that the new variables are used everywhere.

phoddie, 2020-10-30:
@WebSamuraiDaisuke, do you have an update on this issue? Thank you.

phoddie, 2020-11-05:
As this issue has been idle for a couple weeks, I'm going to close it. If the issue persists, please re-open and provide additional details. Thank you.

closed 2020-11-05 by phoddie