Moddable-OpenSource/moddable #479: no permission to delete 5th property on BigInt64Array

dckc, 2020-10-21:
**Build environment:**  Linux
**Target device:** xsnap

**Description**
`BigInt64Array` has a 5th property whose name is a symbol. It cannot be deleted.

SES requires that all non-whitelisted properties on primordials be deletable.

**Steps to Reproduce**

run this as a module, i.e. in strict mode:

```
// surprising property on BigInt64Array
// undeletable even though not frozen
// will be a problem when SES whitelist...

const props = Reflect.ownKeys(BigInt64Array);
print(typeof props[4]);
delete BigInt64Array[props[4]];  // must succeed in order for SES to work. non-whitelist must be deletable
// bug1.js:6: exception: bug1.js: delete ?: no permission (strict mode)!
```

cc @warner @erights @kriskowal @phoddie

phoddie, 2020-10-21:
This behavior is a consequence of an internal slot being made visible to scripts. The mistake is that an internal slot is placed in the wrong position, after a public slot. This occurs for all `TypedArray` constructors, not only `BigInt64Array`. We'll have a fix posted in a little bit.

phoddie, 2020-10-21:
[Fix](https://github.com/Moddable-OpenSource/moddable/commit/748fda9359ea20ec0b4ff224f2cf077f47b6e507) is now available.

dckc, 2020-10-21:
verified.

`print(typeof props[4]);` gives `undefined`. And there's no exception from `delete`.

closed 2020-10-21 by phoddie