Moddable-OpenSource/moddable #481: cannot coerce to instance from 5th property on BigInt64Array

dckc, 2020-10-21:
_IOU some additional details. We tried to find an isolated way to trigger this exception, but we weren't able to separate it from a somewhat complex setup. I'm submitting this issue primarily as a common point of reference._

The code that triggered the `Object: cannot coerce to instance` exception was:

```js
        if (Object(val) !== val) {
```

Our harden code was able to describe `val` as `unknown.BigInt64Array.Symbol()`.

Some very strange things happened. We couldn't print `typeof val` or even `typeof typeof val` in this context.

This diagnostic seems to occur exactly once in the xs sources, in a `default` case in `fxToInstance`:

```
~/projects/moddable$ ack 'cannot coerce to instance'
xs/sources/xsType.c
243:		mxTypeError("cannot coerce to instance");
```

cc @warner @erights @kriskowal @phoddie

dckc, 2020-10-21:
I found a bit more context. The exception happens within...

```js
  c = new CompartmentToUse(endowments, {}, compartmentOptions);
  harden(c.globalThis);
```

https://github.com/Agoric/agoric-sdk/blob/master/packages/import-bundle/src/index.js#L55

phoddie, 2020-10-21:
We don't fully understand this well enough to investigate. Perhaps the fix for #479 helps here too?  If not, we will need additional details.

phoddie, 2020-10-22:
Now that a the problem with the internal slot leaking on TypedArray constructors has been fixed, is this problem still active?

dckc, 2020-10-22:
no, these symptoms have gone too, I guess.

I was hesitant to say so, but I suppose I have verified it.

closed 2020-10-22 by dckc

phoddie, 2020-10-22:
Thanks. If it decides to make a comeback, just reopen this issue with the details.