Moddable-OpenSource/moddable #482: piu/Sound not works in M5stack core2

stc1988, 2020-10-21:
**Version:** October 20, 2020
**Build environment:** Windows
**Target device:** M5stack Core2

**Description**
I try to run "examples/piu/sound" in M5stack Core2, but not play sound.
.
**Steps to Reproduce**
1. Add `"esp32/m5stack_core2": {},` to "platforms" of `examples\piu\sound\manifest.json`
2. Build command is `mcconfig -d -m -r 0 -p esp32/m5stack_core2`
3. Application run.
4. Touch "playSound" Button.

**Expected behavior**
When I touch "playSound" Button, play sound.

**Images**
I took a video. It includes "Other information" behavior.
https://twitter.com/stc1988/status/1318915871343931394

**Other information**
* When I remove startupSound of  `build\devices\esp32\targets\m5stack_core2\manifest.json` like `"startupSound": ""`,  Application plays sound. 
* Addding `"esp32/m5stack" to "examples\piu\sound" works in M5stack.

So I think code of  example application has no problem.

phoddie, 2020-10-22:
i don't have an M5Stack Core2, so i can only guess about the problem. Since you are able to play sound, it seems like a higher level issue.

When the startup sound is used, it leaves the AudioOut instance open afterwards. When Piu Sound plays sound, it creates another audio instance. That has always worked without a problem because only one is active at a time. It seems maybe here having two instances open is causing a conflict, even when one is inactive. (I'm guessing....)

Let's try changing the code that plays the startup sound to close the audio output instance after the sound is complete. Here's the current code:

https://github.com/Moddable-OpenSource/moddable/blob/748fda9359ea20ec0b4ff224f2cf077f47b6e507/build/devices/esp32/targets/m5stack_core2/setup-target.js#L21-L29

```js
  global.speaker = new AudioOut({ streams: 4 });
  if (config.startupSound) {
    speaker.callback = function () {
      this.stop();
      this.close();
      this.done?.();
    };
    speaker.enqueue(0, AudioOut.Samples, new Resource(config.startupSound));
    speaker.enqueue(0, AudioOut.Callback, 0);
    speaker.done = done;
    done = undefined;
    speaker.start();
  }
```

You'll also need to change the call to `done();` at the end of the function to:

```js
  done?.();
```

Let me know how that goes. Thanks for the help!

N.B. I don't consider this a real fix, but an experiment to better understand the problem.

stc1988, 2020-10-22:
I Changed the above code, but below error has occurred. Example application fails to run. 
This code works in M5stack.

> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x40088974  PS      : 0x00060530  A0      : 0x8012ecba  A1      : 0x3ffc2680
> A2      : 0xffffffff  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000000
> A6      : 0xffffffff  A7      : 0x00000600  A8      : 0x80153b40  A9      : 0x3ffc2660
> A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffc3478  A13     : 0x00000000
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c
> EXCVADDR: 0x0000003f  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> ELF file SHA256: d77df35a82081b01c660212993de914d3a6c1630a4d7d3356af657461a32977d
> Backtrace: 0x40088974:0x3ffc2680 0x4012ecb7:0x3ffc26c0 0x40087e9a:0x3ffc2720
> Entering gdb stub now.

I add `debugger;` before  `this.done?.();`, it works as expected.

``` javascript
    speaker.callback = function () {
      this.stop();
      this.close();
      debugger;
      this.done?.();
    };
```
Or adding `debugger;` before  new Sound of example application, it works as expected.

```javascript
debugger;
export default new SoundApp({ sound: new Sound({ path: "bflatmajor.wav" }) }, { displayListLength: 1024, touchCount: 1 });
````
Thefore,  this problem seems that two instances open is causing a conflict.
Does closing AudioOut instance takes time?

phoddie, 2020-10-22:
Thanks for the update. Your observation about the delay caused by the `debugger` statement is helpful. Yes, the task can take a little time to terminate. It is actually longer than I imagined -- around 75 ms.

I've attached a diff that you can apply to [audioOut.c](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/i2s/audioout.c). It has `close` wait until the I2S driver has been uninstalled. It is stable on M5Stack Fire, but you'll have to let me know about the M5Stack core2.

[audiooutpatch.diff.txt](https://github.com/Moddable-OpenSource/moddable/files/5426019/audiooutpatch.diff.txt)

stc1988, 2020-10-22:
I apply a patch to `audioOut.c.`
Displaying application seems to delay a little compared to M5tack., but it works as i expeceted!

I took a [video](https://twitter.com/stc1988/status/1319412469991759873?)

phoddie, 2020-10-22:
Excellent! Thank you.

Yes, because we wait for the sound to finish before completing the setup step, there is a small delay. I think this approach is safer though. I'll think about it a bit more, but I think it is probably safest to apply this change across all the devices. Even on the devices where it is not causing a problem currently, it seems like a fragile state.

stc1988, 2020-10-31:
I confirmed fixed  in October 30, 2020.Thank you for your help. 

Could you add `esp32/m5stack_core2: {},`  and  `esp32/m5stack: {},` to "platforms" of `examples\piu\sound\manifest.json` ?

phoddie, 2020-10-31:
Thanks for verifying the fix. I'll update the example manifest to add the additional device targets.

phoddie, 2020-11-05:
The manifest has been updated as requested.

https://github.com/Moddable-OpenSource/moddable/blob/a4616854e6692daff6fdb8b6d9d7cbb4dffbfdac/examples/piu/sound/manifest.json#L29-L33

stc1988, 2020-11-05:
I confirmed. Thank you!!

closed 2020-11-05 by stc1988