Moddable-OpenSource/moddable #483: heap-buffer-overflow at xs/sources/xsBigInt.c:1354

kvenux, 2020-10-22:
# Build environment:
Ubuntu 16.04
gcc 5.4.0
xst version: 748fda9
build command:
cd /path/to/moddable/xs/makefiles/lin
make
test command: ./xst poc

# Target device:
Desktop Linux

# POC
[xs-new-000093.txt](https://github.com/Moddable-OpenSource/moddable/files/5423531/xs-new-000093.txt)

# Description
Below is the ASAN outputs.

=================================================================
==49977==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60200000e43c at pc 0x0000004cb7de bp 0x7ffc103532b0 sp 0x7ffc103532a0
WRITE of size 4 at 0x60200000e43c thread T0
    #0 0x4cb7dd in fxBigInt_uadd /home/keven/Fuzzing/moddable-new/xs/sources/xsBigInt.c:1354
    #1 0x4ce497 in fxBigIntParse /home/keven/Fuzzing/moddable-new/xs/sources/xsBigInt.c:443
    #2 0x681e35 in fxGetNextNumberE /home/keven/Fuzzing/moddable-new/xs/sources/xsLexical.c:413
    #3 0x68f259 in fxGetNextTokenAux /home/keven/Fuzzing/moddable-new/xs/sources/xsLexical.c:932
    #4 0x69aa57 in fxGetNextToken /home/keven/Fuzzing/moddable-new/xs/sources/xsLexical.c:811
    #5 0x8eb3da in fxParametersBinding /home/keven/Fuzzing/moddable-new/xs/sources/xsSyntaxical.c:3492
    #6 0x8fe02f in fxFunctionExpression /home/keven/Fuzzing/moddable-new/xs/sources/xsSyntaxical.c:2591
    #7 0x8f7413 in fxStatement /home/keven/Fuzzing/moddable-new/xs/sources/xsSyntaxical.c:1193
    #8 0x8fd258 in fxBody /home/keven/Fuzzing/moddable-new/xs/sources/xsSyntaxical.c:1079
    #9 0x90bc81 in fxProgram /home/keven/Fuzzing/moddable-new/xs/sources/xsSyntaxical.c:1065
    #10 0x911981 in fxParserTree /home/keven/Fuzzing/moddable-new/xs/sources/xsTree.c:168
    #11 0x77804c in fxParseScript /home/keven/Fuzzing/moddable-new/xs/sources/xsPlatforms.c:435
    #12 0x852470 in fxRunEval /home/keven/Fuzzing/moddable-new/xs/sources/xsRun.c:4177
    #13 0x80b909 in fxRunID /home/keven/Fuzzing/moddable-new/xs/sources/xsRun.c:3904
    #14 0x850672 in fxRunScript /home/keven/Fuzzing/moddable-new/xs/sources/xsRun.c:4606
    #15 0xa31992 in fxRunProgramFile /home/keven/Fuzzing/moddable-new/xs/tools/xst.c:1398
    #16 0x41902a in main /home/keven/Fuzzing/moddable-new/xs/tools/xst.c:290
    #17 0x7fbdceefc83f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2083f)
    #18 0x41bb98 in _start (/home/keven/Fuzzing/moddable-new/build/bin/lin/debug/xst+0x41bb98)

0x60200000e43c is located 0 bytes to the right of 12-byte region [0x60200000e430,0x60200000e43c)
allocated by thread T0 here:
    #0 0x7fbdcf864602 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98602)
    #1 0x87ee5f in fxNewParserChunk /home/keven/Fuzzing/moddable-new/xs/sources/xsScript.c:126

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/keven/Fuzzing/moddable-new/xs/sources/xsBigInt.c:1354 fxBigInt_uadd
Shadow bytes around the buggy address:
  0x0c047fff9c30: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9c40: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9c50: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9c60: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff9c70: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c047fff9c80: fa fa fa fa fa fa 00[04]fa fa 00 06 fa fa 00 05
  0x0c047fff9c90: fa fa 00 06 fa fa 00 04 fa fa 00 06 fa fa 00 05
  0x0c047fff9ca0: fa fa 00 07 fa fa 00 07 fa fa 00 07 fa fa 00 06
  0x0c047fff9cb0: fa fa 00 04 fa fa 00 07 fa fa 00 04 fa fa 00 03
  0x0c047fff9cc0: fa fa 00 05 fa fa 00 04 fa fa 00 05 fa fa 00 07
  0x0c047fff9cd0: fa fa 00 05 fa fa 00 04 fa fa 00 07 fa fa 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
==49977==ABORTING

phoddie, 2020-10-22:
Can be reduced to this:

```js
eval(1e7 + "nu");
```

phoddie, 2020-11-05:
The source text contained an invalid number but the parser tried to continue.

closed 2020-11-08 by kvenux