Moddable-OpenSource/moddable #485: SEGV at /xs/sources/xsBigInt.c:1182

kvenux, 2020-10-22:
# Build environment:
Ubuntu 16.04
gcc 5.4.0
xst version: 748fda9
build command:
cd /path/to/moddable/xs/makefiles/lin
make
test command: ./xst poc

# Target device:
Desktop Linux

# POC
[xs-new-000028.txt](https://github.com/Moddable-OpenSource/moddable/files/5423620/xs-new-000028.txt)

# Description
Below is the ASAN outputs.

ASAN:SIGSEGV
=================================================================
==88944==ERROR: AddressSanitizer: SEGV on unknown address 0x7f2ff591a8af (pc 0x7f2ff1ecc964 bp 0x7ffd3222eb30 sp 0x7ffd3222e2b8 T0)
    #0 0x7f2ff1ecc963  (/usr/lib/x86_64-linux-gnu/libasan.so.2+0xaa963)
    #1 0x7f2ff1eaee8d in __asan_memmove (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x8ce8d)
    #2 0x4e0116 in memmove /usr/include/x86_64-linux-gnu/bits/string3.h:59
    #3 0x4e0116 in fxBigInt_ulsl1 /home/keven/Fuzzing/moddable-new/xs/sources/xsBigInt.c:1182
    #4 0x4f4514 in fx_BigInt_asUintN /home/keven/Fuzzing/moddable-new/xs/sources/xsBigInt.c:167
    #5 0x7e7d41 in fxRunID /home/keven/Fuzzing/moddable-new/xs/sources/xsRun.c:769
    #6 0x850672 in fxRunScript /home/keven/Fuzzing/moddable-new/xs/sources/xsRun.c:4606
    #7 0xa31992 in fxRunProgramFile /home/keven/Fuzzing/moddable-new/xs/tools/xst.c:1398
    #8 0x41902a in main /home/keven/Fuzzing/moddable-new/xs/tools/xst.c:290
    #9 0x7f2ff155283f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2083f)
    #10 0x41bb98 in _start (/home/keven/Fuzzing/moddable-new/build/bin/lin/debug/xst+0x41bb98)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV ??:0 ??
==88944==ABORTING

phoddie, 2020-11-05:
`BigInt` data are limited to 65535 32-bit unsigned integers but XS did not check.

phoddie, 2020-11-05:
Hey @kvenux, we **really** appreciate your steady stream of reports. Each one of them helps make XS a little more robust and secure for everyone. We'd like to send you a Moddable t-shirt to say thank you. Just drop an email to info@moddable.com with your mailing address and size.

closed 2020-11-08 by kvenux