Moddable-OpenSource/moddable #489: snapshot slows down 10x over 70 trivial iterations

dckc, 2020-10-28:
**Steps to Reproduce**
run `python3 go.py` with `xsnap` in `$PATH`. It runs 70 iterations of...

```
ix += 1;
print(ix);
```

with a restore before and snapshot after each.

Observe roughly linear slow down; the 70th iteration is 10x slower than the 1st: ~0.5sec vs ~0.05sec.

details: https://gist.github.com/dckc/a2b67caae93fbb43c2825019fecd732d

**Expected behavior**
roughly constant runtime, given that each iteration is a constant amount of work.

**Build environment:** linux
xsnap rev e8199b1012981d627a77235bb6737f9deef07103
moddable version: 748fda9359ea20ec0b4ff224f2cf077f47b6e507 + map snapshot fix
```
diff --git a/xs/sources/xsSnapshot.c b/xs/sources/xsSnapshot.c
index 9569d055..d02aec26 100644
--- a/xs/sources/xsSnapshot.c
+++ b/xs/sources/xsSnapshot.c
@@ -760,9 +760,7 @@ txMachine* fxReadSnapshot(txSnapshot* snapshot, txString theName, void* theConte
                        mxThrowIf((*snapshot->read)(snapshot->stream, the->freeHeap, atom.atomSize));
                        the->freeHeap = the->freeHeap + (atom.atomSize / sizeof(txSlot));
        
-                       slot = the->firstHeap;
-                       slot->value.reference = the->freeHeap;
-                       slot++;
+                        slot = the->firstHeap + 1;
                        while (slot < the->freeHeap) {
                                fxReadSlot(the, snapshot, slot, 1);
                                slot++;
```

phoddie, 2020-10-29:
The problem is memory. Basically, on each execution the heap was is increasing by 16 MB. By thee 70th iteration it was over 1 GB. The allocation is taking time at launch. The performance of the execution of the scripts is more-or-less constant as expected.

The allocation shouldn't grow on each launch. That was a bug. It grew very quickly because the incremental chunk heap grow size is 16 MB to accommodate the restively large strings your module loader creates.

The fix is easy. Between these two lines...

https://github.com/Moddable-OpenSource/moddable/blob/748fda9359ea20ec0b4ff224f2cf077f47b6e507/xs/sources/xsMemory.c#L401-L402

...add:

```c
theSize -= sizeof(txBlock);
```

We'll include this in the next Moddable SDK update.

dckc, 2020-10-29:
I'm not able to confirm.

I added that line and rebuilt xsnap but I see no change in the symptoms. I wonder what I'm doing different from what you're doing.

I would appreciate a copy of the output you get from `go.py`.

phoddie, 2020-10-29:
Summary: I just tried this on macOS. It works as expected.

**Details**

On my MacBook Air, I see about 22.5 iterations per second.

Unfortunately, the Python script doesn't work for me and debugging Python isn't something I can do. Instead, I used a simple batch (for macOS):

```
rm -f state0.xsnap
rm -f state1.xsnap

xsnap -s perf-start.js -w state0.xsnap

while true
do
	xsnap -s perf-iter.js -r state0.xsnap -w state1.xsnap
	xsnap -s perf-iter.js -r state1.xsnap -w state0.xsnap
done
```

The test as given doesn't work as intended. When invoking xsnap with `-m` the source text is loaded as a module. Modules are loaded only once, so subsequent attempts to load it in the same execution environment (e.g. the loaded snapshot) do nothing. Consequently, the counter doesn't increment beyond 1. Use `-s` to execute the source text as a script as the above batch does. (This same mistake was noted in #488)

To see if the problem you observe is the same memory growth we observed, you can enable memory reporting by setting `mxReport` to 1 at line 41 of xsMemory.c. After rebuilding xsnap, you should observe stable output for each iteration as below.

```
$ ./test.bat 
# Chunk allocation: reserved 16777216 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
# Chunk collection: reserved 16777216 used 113688 peak 114368 bytes
# Slot collection: reserved 33554432 used 120608 peak 120768 bytes 0
# Chunk allocation: reserved 16777216 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
1
# Chunk collection: reserved 16777216 used 113752 peak 108 bytes
# Slot collection: reserved 33554432 used 120640 peak 192 bytes 0
# Chunk allocation: reserved 16777216 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
2
# Chunk collection: reserved 16777216 used 113752 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 16777216 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
3
# Chunk collection: reserved 16777216 used 113752 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 16777216 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
4
...
```

 If the bug is active, the "Chunk allocation" grows with each iteration.

dckc, 2020-10-29:
OK, thanks for these details.

(Yes, I realized the `-m` issue right after I reported this. I checked without `-m` at the time and the symptoms were the same so I didn't change anything for fear of muddying the waters.)

dckc, 2020-10-29:
The bug seems to be active here even with `theSize -= sizeof(txBlock);`:

```
$ ./test.bat
# Chunk allocation: reserved 16777248 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
# Chunk collection: reserved 16777248 used 113728 peak 114408 bytes
# Slot collection: reserved 33554432 used 120608 peak 120768 bytes 0
# Chunk allocation: reserved 33554464 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
1
# Chunk collection: reserved 33554464 used 113836 peak 152 bytes
# Slot collection: reserved 33554432 used 120640 peak 192 bytes 0
# Chunk allocation: reserved 50331680 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
2
# Chunk collection: reserved 50331680 used 113836 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 67108896 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
3
# Chunk collection: reserved 67108896 used 113836 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 83886112 used 0 peak 0 bytes
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
...
```

In case it's useful, I changed the `fxCall` to report `theSize` as well:

```
$ ./test.bat
# Chunk allocation: reserved 16777248 used 0 peak 0 bytes theSize: 16777216
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
# Chunk collection: reserved 16777248 used 113728 peak 114408 bytes
# Slot collection: reserved 33554432 used 120608 peak 120768 bytes 0
# Chunk allocation: reserved 33554464 used 0 peak 0 bytes theSize: 33554432
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
1
# Chunk collection: reserved 33554464 used 113836 peak 152 bytes
# Slot collection: reserved 33554432 used 120640 peak 192 bytes 0
# Chunk allocation: reserved 50331680 used 0 peak 0 bytes theSize: 50331648
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
2
# Chunk collection: reserved 50331680 used 113836 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 67108896 used 0 peak 0 bytes theSize: 67108864
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
3
# Chunk collection: reserved 67108896 used 113836 peak 44 bytes
# Slot collection: reserved 33554432 used 120640 peak 160 bytes 0
# Chunk allocation: reserved 83886112 used 0 peak 0 bytes theSize: 83886080
# Slot allocation: reserved 33554432 used 0 peak 0 bytes
...
```

dckc, 2020-10-29:
Where am I supposed to put `theSize -= sizeof(txBlock);` again? I don't see how it can have any effect after line 401. When I put it _before_ line 401, I do see the symptoms subside.

Perhaps you meant "before these two lines" rather than "between these two lines"?

phoddie, 2020-10-30:
Ah, yes. My mistake. It needs to be before those two lines, not between them. Apologies.

dckc, 2020-10-30:
It's all good. Especially the mxReport clue.

Do we close this now or after the fix is available in the public branch?

phoddie, 2020-10-30:
We plan to update this repository today, so either is fine.

phoddie, 2020-11-05:
Closing now that the fix is committed here.

closed 2020-11-05 by phoddie