Moddable-OpenSource/moddable #491: Can not get the SDK running on Windows 10

MarcelisProgramming, 2020-10-31:
Hello,
I can't get the SDK running. First of all, the instructions are not clear to me. E.g. here in point 4 I have to set the MODDABLE environment variable. But where do I do that? In the Github prompt, or in the Windows Powershell?
Thanks!
Kind regards,
Marcel
[image]

phoddie, 2020-10-31:
We are happy to help try to resolve the problem. For us to be able to help, you will need to provide specific information about the problems you are having. What is it that is unclear to you?

Many people have successfully set-up the Moddable SDK on Windows. It can be done.

andycarle, 2020-10-31:
Hi @MarcelisProgramming,

(Looking back through your edits to try to understand the sticking point here.) The Windows environment variables should be set through the Control Panel app. Please follow [this guide](https://www.architectryan.com/2018/08/31/how-to-change-environment-variables-on-windows-10/) if you are unfamiliar with this process.

And thank you for the feedback on the instructions—I have clarified the Windows environment variables sections in an upcoming revision of our Getting Started documentation. 

Cheers!
&nbsp;- Andy

MarcelisProgramming, 2020-10-31:
Hi,
Everything is going now, except that I can't get the compiling for ESP32 running.
I do exactly as the instructions say, but keep getting this error while compiling, and everything stops.
Hope you have some tips.
Kind regards,
Marcel
[image]

MarcelisProgramming, 2020-11-01:
Hello,
I got it working now.
Thank you guys!
Kind regards,
Marcel :-)

andycarle, 2020-11-01:
Hi @MarcelisProgramming,

Great! I'm glad to hear that you got everything fixed up. Out of curiosity: what was the issue that was causing the `0x2331` error you posted above? 

&nbsp;- Andy

MarcelisProgramming, 2020-11-02:
Hi Andy,
Something must have gone wrong while installing the ESP-IDF Tools Installer. Maybe the downloaded file was corrupted or something. After I repeated the installationproces with a new downloaded file, it worked well.
Marcel

andycarle, 2020-11-03:
Great; thank you for the info. 

I'm going to go ahead and close this out. Feel free to re-open the issue if you have any further problems!

&nbsp;- Andy

closed 2020-11-03 by andycarle