Moddable-OpenSource/moddable #494: How to add user libraries during testing?

ariznaf, 2020-11-15:
I am new to moddable SDK.

I am trying to get to make some development in javascript and moddable skd in a esp32.
To begin with I will try to develope a wheather station that puslishes its data (temperature, humidity and preassure) using mqtt.

I have made some progress, but now I am stuck in how to convert my code to a user library.
I hava made some classes and they work as long as they reside in a directory and I put that directory in modules section of the manifest.
```
	"modules": {
		"*": [
			"./app",
			"../ArAlLib/*"
		]
```
there is a file name **miclass.js** exporting a class:
```
Class MyClass {
}
export default MyClass
```
and if I import it in my main.js it works
` import MyClass from "myclass"`

But now I would like to organize classes in order to have a reusable library and my own root namespace (Aral) with a hierachy (like **Aral/util/myclass**).
I have moved myclass.js to a **util** subdir.
I have changes the import statement in **main.js** to:
`import myClass from "Aral/util/myclass"`

But it does not work.
I have tried to to change the manifest like this, with no success:
```
	"modules": {
		"*": [
			"./app",
		],
                "Aral/util": "../AralLib/util/*"
```

How should I write the manifest in order to be able to import my libs of classes from a structured tree?

phoddie, 2020-11-16:
> To begin with I will try to develop a weather station that puslishes its data (temperature, humidity and pressure) using mqtt.

Sounds like a great project to learn how to put all the pieces together.

Let's make some changes to [$MODDABLE/examples/helloworld](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/helloworld) to walk through how to do this.

First, add a directory named `utils` at $MODDABLE/examples/helloworld/utils with three files named `a.js`, `b.js`, and `c.js`. The contents of each will be as follows but change the letter in the `trace` to match the file name.

```js
export default function() {
	trace("a\n");
};
```

To add these to the project we'll modify the manifest to look like this:

```json
	"include": "$(MODDABLE)/examples/manifest_base.json",
	"modules": {
		"*": "./main",
		"utility/*": "./utils/*"
	}
}
```

This causes all the JavaScript files in the `utils` directory to be included in the project. They will be available as a module with a path starting with `utility/` and ending with the module name. So, for example, `a.js` will be available to import as a module from `utility/a`.

Notice that the right side of the expression, `./utils/*` specifies the directory path of the source code, and the left side, `utility/*`, specifies what you will use to import the module in your code (the module specifier). Often they are they same, but in this example they are different to show the ability of the manifest to perform remapping.

Here's the modified version of `main.js` that imports all three modules and invokes them:

```js
import a from "utility/a";
import b from "utility/b";
import c from "utility/c";

a();
b();
c();
```

When run, this traces the following to the xsbug debug log:

```
a
b
c
```

Note that the `*` wildcard also works with the `preload` element of the manifest:

```json
	"preload": "utility/*"
```

Preload is an advanced feature that saves memory and speeds start-up. The details are available [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md).

ariznaf, 2020-11-16:
Thank you, now it works.

But it seems not to be recursive.

I mean, if I have a dir mylibs with subdirs util and tools with this structure:

```
mylib/
        util/
             util1.js
       tools/
             tool1.js
 ```

I woul like to use imports like this

```
import util1 from "mylib/util/util1";
import tool1 from "mylib/util/tool1";
```

I have tried this manifest:
```
{
	"include": "$(MODDABLE)/examples/manifest_base.json",
	"modules": {
		"*": "./main",
		"mylib/*": "../mylib/*"
         }
}
```

But it does not seem to work, it does not make recursive searches of dirs.

So I have tried this other schema

```
{
	"include": "$(MODDABLE)/examples/manifest_base.json",
	"modules": {
		"*": "./main",
		"mylib/util/*": "../mylib/util/*",
		"mylib/tools/*": "../mylib/tools/*"
         }
}
```

Which does work.

phoddie, 2020-11-16:
Correct. The wildcard applies to the directory specified. It is not recursive.

ariznaf, 2020-11-16:
Thank you, I will close the topic, as the question is solved.

closed 2020-11-16 by ariznaf