Moddable-OpenSource/moddable #495: Is loadmod ready for use?

wilberforce, 2020-11-16:
I'm trying to follow: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/mods.md

and files from here:
https://github.com/Moddable-OpenSource/iot-product-dev-book/tree/master/loadmod

The UPLOAD_SPEED and UPLOAD_PORT (defaults to COM8) does not get read from the board default manifest by `mcrun` like `mcconfig` does.

`mcrun -d -m -p esp32/m5stick_c`

This defaults to the moddable 2 speed - and when I manually try to run using the 1500000 Buad rate - I seen nothing in xsbug and the script does not finish.

a. Is there a compete example with a host and a mod?

b. Does this work with the simulator? Is there anywhere the mods are installed for the simulator?

Thanks

phoddie, 2020-11-16:
Mods work well. You'll hearing more about those from us soon.

That said, I haven't tried them (recently, al least) on a device that uses a baud rate of 1500000.

There are lots of examples [here](https://github.com/Moddable-OpenSource/iot-product-dev-book). I suggest starting with the Chapter 10 host and graph example. It works on devices (Moddable One, Moddable Two, M5Stack Fire) and the simulator.

[image]

wilberforce, 2020-11-16:
Thanks.

here is a PR for the `$DEBUGGER_SPEED`

https://github.com/Moddable-OpenSource/moddable/pull/496

closed 2020-11-17 by wilberforce