Moddable-OpenSource/moddable #497: Implement tree import of modules.

ariznaf, 2020-11-17:
It seems that you cannot use a path with more than one level (nested path) in the modules section of a manifest.

I mean something like:

´´´
{
    "modules": {
        "*": "./app",
        "aral/util/*": "../arallib/util/*"
    }
}
```

In order to be able to use imports  in code like:

` import Tool from "aral/util/tool"`

If you try to use it, when you run mcrun you get an error (not very useful error, as it does not provide any hint about where it comes from).
```
d:\...\pruebasModdable\tests>mcrun -d -m -p esp32 ssid="myssid" password="***"
### Error: No such file or directory
´´´
It seems that mcrun does not interpret paths with more than one level.

A workaround (provided by @phoddie) it to use this instead:
```
{
    "modules": {
        "*": "./app",
        "aral_util/*": "../arallib/util/*"
    }
}
```
` import Tool from "aral_util/too"`

But it is ugly not to have standard import paths.

phoddie, 2020-11-17:
The tools now support multi-part paths. To use that, you will need to update the Moddable SDK and rebuild tools. The [Keeping Up To Date](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Keeping%20Up%20To%20Date.md) document explains how to do that.

ariznaf, 2020-11-18:
Woww. that is eficiency.

I will see if I can upgrade this afternoon..

By the way ... does it import all the subtree or just one level?

I mean if I have aral/util and aral/componentes, would I need to create 
```
{
    "modules": {
        "*": "./app",
        "aral/components/*": "../arallib/components/*"
        "aral/util/*": "../arallib/util/*"
    }
}
```

Or would it be enough with ...
```
{
    "modules": {
        "*": "./app",
        "aral/*": "../arallib/*"
    }
}
```

phoddie, 2020-11-18:
Thanks for giving it a try.

FYI - The module wildcard applies to one level. It is not recursive. While it would be convenient to have recursive imports for very large projects, properly supporting them is not trivial and so we don't intend to support recursive imports at this time.

ariznaf, 2020-11-19:
OK, it is no a problem to import each subdir. It may be even convenient (so you don't forget what you are exactly using).

I will try to update my moddable sdk install and a test it, and will feed back (I could not do it yesterday):

ariznaf, 2020-11-19:
It did not work for me.

I have changed the manifest to this:
```
{
	"modules": {
		"*": [
			"./app"
		],
		"aral/util/*": "../arallib/util/*",
		"aral/components/*": "../arallib/components/*"
	}
}
```

and the import statement to this:
```
import Color from "aral/util/rgbcolor";
```

And got the eror again:
`### Error: No such file or directory`

ariznaf, 2020-12-05:
Any update on this?

I have tried it but it does not work yet.
May be there is somthing broken in my install?

phoddie, 2020-12-05:
The feature is working as expected. Please see the example [here](https://gist.github.com/phoddie/714a099644a307ace1ebc3fccbe6f8a9).

ariznaf, 2020-12-07:
I am getting mad with this.

I have tried to test the example you provided, but I continue getting an error:
[image]

I have checked every thing I could think of.
I have reinstalled moddable sdk and the espressif tools.

May be it is a windows bug?
I have made a git proyecto in github here:
https://github.com/ariznaf/pruebasMoodable/tree/main/treeImport-issue497

I have tried your example using the host integrated in one file (in the app directory as in your example) and creating a host (extracted from ch1 of your book it is in host directory) and a separate app to run with mcrun (in the test directory).

Neither of them work.
Any idea or any test I can try to fix the problem?

wilberforce, 2020-12-07:
@ariznaf 

It looks like the `mcconfig` can't be found. Have you updated the `PATH` environment variable? If you have you might need to start another cmd window to get that setting.

ariznaf, 2020-12-07:
mcconfig?

mcconfig runs and flashes the firmware. I have run and flashed other software from the examples and the iot bood.

It is only when I try to use the import in multi level tree when there is an error, and it is thrown after mcconfig is launched and running, is the mcconfig process the one that trows the error.

phoddie, 2020-12-07:
Perhaps the change to `mcconfig` is working on macOS but not Windows. That would explain the difference in our results. Path handling is different on Windows. We'll investigate  that possibility.

ariznaf, 2020-12-07:
Yes, it seems it is platform related.

I suspect that the function you use may be using to detect the / is changing it by \ in windows or something like that (interpreting the string as a directory).

@cmidgley has tested it with the same results and he is using windows too.

I am stuck with this problem and the check module not been loaded.

phoddie, 2020-12-07:
This issue is about multi-part module specifiers. The `check` module does not use that and so is unrelated to the topic of this issue.

ariznaf, 2020-12-07:
Yes, you are right it seems another problem (I am sure that the problem is my manifest, in that case).

phoddie, 2020-12-08:
A [fix](https://github.com/Moddable-OpenSource/moddable/commit/b93c6fd5b36c085c738ff3345c157cf037dee5d7) has been committed for this on Windows.

To try it:

- Update Moddable SDK sources
- Rebuild Moddable SDK tools
- clean your test project
- build your test project

ariznaf, 2020-12-08:
I have upgraded moddable and tried.
It now works OK.

Thank you very much for the fix. I close the issue now, as it is solved (I am not sure if I should close or I should let you that task).

closed 2020-12-08 by ariznaf

phoddie, 2020-12-08:
Wahoo!

Thank you for your patience. The cross-platform differences sometimes obscure problems. 

Once you have confirmed a problem is resolved, it is helpful to close the issue. Thanks for doing that.