Moddable-OpenSource/moddable #498: RTC "set seconds" sets late date

stc1988, 2020-11-17:
**Version:** November 16, 2020
**Build environment:** Windows
**Target device:** M5stick_c

**Description**

Value of get seconds after set seconds is late date

**Expected behavior**
The value of set seconds and get seconds is same value

**Images**
[image]

trace log before `set rtc.seconds` is `Nov 17 2020`, but displaying date of  `get rtc.seconds` is `Nov 19 2020`

**Steps to Reproduce**
1. build `examples/drivers/m5stickc-rtc`
build command is mcconfig -d -m -p esp32/m5stick_c

2. Press button B to set seconds to RTC.

phoddie, 2020-11-17:
@wilberforce - I think this is your code. Would you take a look? (I don't have an M5 Stick C to try.) Thanks.

wilberforce, 2020-11-19:
Sure. I’ll take a look. It looks like the year offset is wrong.

Will use this as a reference:

https://github.com/tanakamasayuki/I2C_BM8563

phoddie, 2020-11-19:
Cool, thanks.

stc1988, 2020-11-24:
I added logs before `_setDate` and  after `_getDate` in `modules\drivers\rtc\rtc.js`

```javascript
set seconds(secs) {
	...
	trace(`[setDate]${date.month} ${date.date} ${date.year}\n`);
	this._setDate(date);
}
```

```` javascript
get seconds() {
	let date = this._getDate();
	trace(`[getDate]${date.month} ${date.date} ${date.year}\n`);
````

this is log when button B pressed
> Set time: Tue Nov 24 2020 03:55:11 GMT-0900
> 1606222511.893
> [setDate]10 26 2020
> [getDate]10 26 2020

Values of `setDate` and `getDate` are the same vaue.
So `bm8563.js` works as exepeted, `set seconds` in `rtc.js` seems to have some issues

wilberforce, 2020-11-24:
@stc1988 
You discovered what I did and had not yet publiashed.
 There seems to be a day adjustment in rtc.js, I have simplified these to use javascript time functions, and it behaves as expected:

```

	get seconds() {
		let date = this._getDate();
		let when = new Date(date.year, date.month, date.date, date.hours, date.minutes, date.seconds, 0);
		return when.getTime() / 1000;
	}

	set seconds(secs) {
		let d = new Date(secs * 1000);
		let date = {
			date: d.getDate(),
			month: d.getMonth(),
			year: d.getFullYear(),
			dow: d.getDay(),
			hours: d.getHours(),
			minutes: d.getMinutes(),
			seconds: d.getSeconds()
		};
		this._setDate(date);
	}
```

wilberforce, 2020-11-24:
https://github.com/Moddable-OpenSource/moddable/pull/501

stc1988, 2020-11-24:
This year is leap year. Day adjustment problem is related to leap  year. 

Anyway, Using javascript time function is nice idea.
I will check some cases with your fix.

stc1988, 2020-11-25:
@wilberforce 
I confirmed some cases  including leap day. It works as expected.
Thanks.

stc1988, 2020-11-28:
I confirmed it works again with version `November 27, 2020`.

closed 2020-11-28 by stc1988