Moddable-OpenSource/moddable #499: Is there an e-ink display with touch screen I could use?

ariznaf, 2020-11-23:
I am looking for an e-ink display with touche screen I could use for my developement in esp32.

I would prefer a color display, not a real rgb color display, but there are some that are BN and can display another color like yellow or red.
That would be enough to show alarms or display notification messages in another color.
Something like this (but does not have a touch screen).

https://www.amazon.es/Waveshare-Resolution-Three-Color-Controller-Interface/dp/B079RKRZQL/ref=sr_1_5?dchild=1&keywords=eink+color+display&qid=1606118622&sr=8-5

A resolutions about 300x200 would be enough.

Is there any that is compatible with moddable and piu?

Moddable two has eink display, but 1 bit bn, no touch screen and is a esp8266 board.

Moddable one is good, but it is no eink, and uses esp8266.

Moddable three is not e-ink and is too big, probably.

I am lookin for e-ink in order to show temperature and other data with minimal energy consumption.

Moddable three with e-ink would be perfect, I think.

I am using a esp32-wrooom devkit C board for development.

phoddie, 2020-11-23:
This question is off-topic. It is not about an issue or development question using the Moddable SDK. It is a request for advice selecting a hardware component.

Touch screens on ePaper display are rare.

FWIW - Moddable Three has an ePaper display, without touch, connected  to an ESP8266, which supports Piu and Wi-FI.  If you want to experiment with ePaper and don't want to embark on writing a driver to support a new panel, it is a good place to start.

closed 2020-11-23 by phoddie

ariznaf, 2020-11-23:
Well, yes may be I did not ask the correct questions, sorry.

It is not the exact model what I am trying to know, but the driver compatibility you need in order to get it compatible with Piu without having to write your own drivers.

I could not find info about that, what are the drives compatible with moddable.

Another question is if Piu is compatible with this kind of color display where you only have 3 colors: black, white and yellow (or red). May be you can use them as a three channel display 1 bit depth each?

As I said, moddable three has an esp8266 and it is only Black and white.

May be moddable tree is more appropiate for me for trying (even not eink, and thus it will consume more).

Thank you for your anwser.

phoddie, 2020-11-23:
Detailed information on supported displays together with videos, links to documentation, and wiring diagrams is available [here](https://www.moddable.com/display).

> As I said, moddable three has an esp8266 and it is only Black and white.

You said several things which were not entirely accurate. Here's what you said:

> Moddable **two** has eink display, but 1 bit bn, no touch screen and is a esp8266 board.

Moddable Two does has aa touch screen, ESP32, and an LCD display. And:

> Moddable three is not e-ink and is too big, probably.

Moddable Three uses the same display technology as what i commercial called eInk.. It is also our smallest board.

> May be moddable tree is more appropiate for me for trying (even not eink, and thus it will consume more).

Moddable Three uses the same display technology commonly referred to as eInk.

ariznaf, 2020-11-23:
Sorry, I have switched models.
 Two is the one that uses LCD, ESP32 and touch screen. It seems perfect except for not being e-ink (yes I know it is not easy to find an e-ink panel with touch capacities).

Three is e-ink but just BN and with esp8266.

Thanks for the info.