Moddable-OpenSource/moddable #500: Simulators improvement.

ariznaf, 2020-11-23:
Right now simulators do not implement hardware simulation (digital inputs or outputs, for example).

I understand that implementing a complete hardware simualation by software (alá Proteus) is difficult and time consuming.

But there is an easier alternative that perhaps can be done: using an external microcontroller connected to the PC that is running the simulation via usb.

You cuold flash a special firmware in the micrcontroller to provide the hardware inputs/outpust, and the simulator would send the commands to the MC via usb.

So you could use the simulator to develope the interface in the PC, which is not so tedious as the compile/flashing cycles directly to the microchip.
In applications that are not real time and with inputs and outpus that change not too quick, you could do the complete development in the simulator.

Later you could flash the app to the chip to do final tests without changing the code, as the sim would implement the input/output api sending them directly to the chip.

I2C and other protocols may be implemented that way too.

I have seen that solution previosly developing in C for a micro PIC 32 bits, and it makes development easier.

phoddie, 2020-11-23:
Yes, the Firmata protocol as used by Johnny-Five can be used for that sort of thing.  An implementation of Firmata part of the eModdable SDK. still, It is a non-trivial project to fully realize what you propose. If someone wants to do that work, we would b happy to review the a pull request.

closed 2020-11-23 by phoddie

ariznaf, 2020-11-23:
Great to know there is one part of the equation already available (the usb to hardware firmware).

May be other with more knowledge than me finds the idea usefull and offers to help.
Thank you.