Moddable-OpenSource/moddable #502: mcconfig does not allow for dst on timeset

wilberforce, 2020-11-24:
**Build environment:** Windows
**Target device:** ESP32

**Description**
a. On the windows target, the correct timezone for New Zealand is set (-13)
For the ESP32 it is setting incorrect to -12

b. For the windows target Time.tz and Time.dst are `undefined`

c. SNTP also seems to be be affected - so looks like it's the time setting for the correct offset that is the issue

**Steps to Reproduce**

main.js:

```
import Time from "time";

let d = new Date();
trace(`Set time: ${d.getTime()} ${d.toString()} tz:${Time.timezone} dst:${Time.dst}\n`);
let seconds = d.getTime() / 1000;
let read = seconds;
let s = new Date(read * 1000);
trace(`Get time: ${read} ${s.toString()}\n`)
```

manifest.json:

```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_net.json"
	],
	"modules": {
		"*": [
			"./main"
		]
	}
}
```

**Expected behavior**

Windows:

```
Set time: 1606251468556 Wed Nov 25 2020 09:57:48 GMT+1300 tz:undefined dst:undefined
Get time: 1606251468.556 Wed Nov 25 2020 09:57:48 GMT+1300
```

ESP32:

```
Set time: 1606251628313 Tue Nov 24 2020 09:00:28 GMT-1200 tz:-46800 dst:3600
Get time: 1606251628.313 Tue Nov 24 2020 09:00:28 GMT-1200
```

What you expect to happen

Dst to be reflected on the ESP32 target.
Time.dst and Time.tz to be defined on windows (linux/mac) target

phoddie, 2020-11-24:
Sorry, I'm a little mixed-up. The report says the target is ESP32 but there's a lot of information about Windows.

The simulator on Windows does not implement the `Time` status methods -- they are all [stubs](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/base/time/win/modTime.c). That's consistent with your note that they return `undefined`.

I'm not sure if/how that is related to the timezone being off by an hour on ESP32.

When using xsbug, the time, timezone, and daylight savings values are set automatically for convenience. Here's the ESP32 code that receives those over the debugging connection:

https://github.com/Moddable-OpenSource/moddable/blob/4fd70036dd0b119ed49ac2b7aead075689ae597e/xs/platforms/esp/xsPlatform.c#L1189-L1194

My guess is that the daylight savings offset is incorrect somehow. That often shows up as an "off by an hour" mistake. If that offset is incorrect it is either  because the computer is sending an incorrect offset or the device is applying the offset incorrectly.

The Moddable SDK on ESP32 applies the daylight savings offset in the `updateTZ` function here:

https://github.com/Moddable-OpenSource/moddable/blob/4fd70036dd0b119ed49ac2b7aead075689ae597e/xs/platforms/esp/xsHost.c#L691-L692

The daylight savings offset and timezone are sent by `serial2xsbug`, which owns the debugging connection. That is done in `fxSetTime`. The relevant code for Windows is here:

https://github.com/Moddable-OpenSource/moddable/blob/40162325723fa17b1b64158aee645e46ef16ab2d/tools/serial2xsbug/serial2xsbug.c#L658-L668

I'm hoping that this information will help you to isolate the problem further. Let me know how I can help (I don't have a Windows machine though!).

wilberforce, 2020-11-24:
@phoddie 
Sorry for the confusion. I added windows as a reference.

For the windows target and chome this is the result for `new Date()`

```
new Date(1606251628313);
Wed Nov 25 2020 10:00:28 GMT+1300 (New Zealand Daylight Time)
```

The day 25 and time are correct local time.

For the ESP 32 target:

```
Set time: 1606251628313 Tue Nov 24 2020 09:00:28 GMT-1200 tz:-46800 dst:3600
```

it's showing `GMT -1200` rather than the expected `GMT +1300`.

So it appear the time setting it setting GTM, but not allowing for the correct tz offset:
`-46800=-12*3600 - 1 * 3600`

phoddie, 2020-11-25:
Let's figure out if the ESP32 is receiving the correct time. The time is sent from serial2xsbug in `fxSetTime`

phoddie, 2020-11-25:
Let's see if the Windows machine is sending the correct time to the ESP32. That will help focus things. The time is sent in `fxSetTime`. Before this line:

https://github.com/Moddable-OpenSource/moddable/blob/4fd70036dd0b119ed49ac2b7aead075689ae597e/tools/serial2xsbug/serial2xsbug.c#L684

...let's add this:

```c
	fprintf(stderr, "### set time: %d time, %d gmt, %d dst\n", (int)time,  (int)gmt, (int)dst);
```

Rebuild the Moddable SDK tools and then reinstall your project. You should see the printf in the console as the project starts running. The question is... are the numbers output correct?

wilberforce, 2020-11-25:
console:

```
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
### set time: 1606277921 time, -46800 gmt, 3600 dst
```

Xsbug:
```
Set time: 1606277921314 Tue Nov 24 2020 16:18:41 GMT-1200 tz:-46800 dst:3600
Get time: 1606277921.314 Tue Nov 24 2020 16:18:41 GMT-1200
```

Chrome:
```
new Date(1606277921000)
Wed Nov 25 2020 17:18:41 GMT+1300 (New Zealand Daylight Time)
````

phoddie, 2020-11-25:
Good news: it looks like the ESP32 is using all data provided by your computer. 

Bad news: the GST from computer looks wrong. The GST value is taken from the timezone returned by `_ftime`. The documentation for the timezone says:

> Difference in minutes, moving westward, between UTC and local time

The [example](https://docs.microsoft.com/en-us/cpp/c-runtime-library/reference/ftime-ftime32-ftime64?view=msvc-160) shows that as a positive number. But, your debug output shows it as negative.

Maybe you can sort this out with some experimentation since you have both the Windows computer and the timezone? A good place to start could be to also printf the raw values returned by `_ftime`:

```c
	fprintf(stderr, "### _ftime: %d time, %d timezone, %d dstflag\n", (int)tb.time,  (int)tb.timezone, (int)tb.dstflag);
```

phoddie, 2020-11-25:
I was too tired last night. And timezones always confuse me. Bad combination.

This morning I set my Mac to the New Zealand Timezone. The trace in `fxSetTime` xsbug shows:

```
### set time: 1606326014 time, 43200 gmt, 3600 dst
```

And, the JavaScript date object on ESP32 displays the correct time:

```
Thu Nov 26 2020 06:40:14 GMT+1300
```

So, the bug is in how serial2xsbug interprets the result of `_ftime` on Windows.

On the macOS data above GMT is +12 hours (60 * 60 * 12). That value does not have the DST offset applied.  Windows is currently using -46800 for GMT. We need to get Windows to match macOS (and Linux).

To test any change to Windows, you'll need the GMT for some other time zones on macOS. California (PST) is -28800, Brussels (CET) is 3600, and Nagpur (IST) is 19800 (half hour offset!).

wilberforce, 2020-11-26:
Here is the output:

```
### _ftime: 1606375460 time, -720 timezone, 1 dstflag
### set time: 1606375460 time, -46800 gmt, 3600 dst
```

wilberforce, 2020-11-26:
https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/sntp/main.js

##Windows:

```
Received time 1606375797.
Thu Nov 26 2020 20:29:58 GMT+1300
Thu Nov 26 2020 20:29:59 GMT+1300
Thu Nov 26 2020 20:30:00 GMT+1300
```

##esp32 - no network:
```
Launching app. Type Ctrl-C twice after debugging app to close serial2xsbug...
### _ftime: 1606375941 time, -720 timezone, 1 dstflag
### set time: 1606375941 time, -46800 gmt, 3600 dst
```
No Wi-Fi SSID
Failed: cannot resolve 0.pool.ntp.org
Failed: cannot resolve 1.pool.ntp.org
Failed: cannot resolve 2.pool.ntp.org
Failed: cannot resolve 3.pool.ntp.org
Wed Nov 25 2020 19:32:22 GMT-1200
Wed Nov 25 2020 19:32:23 GMT-1200
```

##esp - with network:
```
Type Ctrl-C twice after debugging app.
### _ftime: 1606376730 time, -720 timezone, 1 dstflag
### set time: 1606376730 time, -46800 gmt, 3600 dst
```

```
Wi-Fi connected to "williams"
IP address 192.168.15.36
Received time 1606376733.
Wed Nov 25 2020 19:45:33 GMT-1200
Wed Nov 25 2020 19:45:34 GMT-1200
```

Chrome:
```
new Date( 1606376733 * 1000 )
Thu Nov 26 2020 20:45:33 GMT+1300 (New Zealand Daylight Time)
```

So both `serial2xsbug` and `sntp` return the correct values, I think it's the esp32 implemention of `Time.set(value);` that is the issue.

wilberforce, 2020-11-26:
It looks like this is the issue:

https://github.com/Moddable-OpenSource/moddable/blob/public/xs/platforms/esp/xsHost.c#L655-L663

```
settimeofday(&tv, NULL);		//@@ implementation doesn't use timezone yet....
```

phoddie, 2020-11-26:
I don't believe that is correct. As noted above, the time and timezone are working fine on ESP32. The comment on `settimeofday` is obsolete and the initial values of `gTimeZoneOffset` and `gDaylightSavings` are initialized by the debugger connection (when present) and may be set using the `Time.timezone` and `Time.dst`.

I don't believe the SNTP information is relevant. It sets the time but not the timez one or dst. If the time zone is set incorrectly, SNTP will not repair that. You can try this simple experiment on ESP32.  Add the two lines below to the sntp example. They set the time zone and daylight savings to the expected values for New Zealand DST.

```
Time.timezone = 43200
Time.dst = 3600;
```

The output is as expected:

```
Wi-Fi connected to "(my Wi-Fi network)"
IP address 192.168.1.33
Received time 1606406552.
Fri Nov 27 2020 05:02:32 GMT+1300
```

The evidence I've seen points to the incorrect timezone being sent by the Windows computer to the ESP32.  

It would really help if you could provide the output of `_ftime` on Windows from the California, Brussels, and Nagpur time zones I noted above. With those four data points it should be possible to figure out how to correct the timezone calculation in `serial2xsbug`.

wilberforce, 2020-11-26:
@phoddie 

I have confirmed this works:

```
Time.timezone = 43200
Time.dst = 3600;
```
```
IP address 192.168.15.36
Received time 1606417231.
Fri Nov 27 2020 08:00:31 GMT+1300
```

Here are the time zones as requested:

NZ: -12
```### _ftime: 1606417274 time, -720 timezone, 1 dstflag
### set time: 1606417274 time, -46800 gmt, 3600 dst
```

California: +8
```### _ftime: 1606417556 time, 480 timezone, 0 dstflag
### set time: 1606417556 time, 28800 gmt, 0 dst
```

Brussels: -1
```
### _ftime: 1606417663 time, -60 timezone, 0 dstflag
### set time: 1606417663 time, -3600 gmt, 0 dst
```

Nagpur:
Can't find in windows

Chatham Islands ( UTC+12:45)
```
### _ftime: 1606418104 time, -765 timezone, 1 dstflag
### set time: 1606418104 time, -49500 gmt, 3600 dst
```

phoddie, 2020-11-29:
Here's the data we have in common (I can't set Chatham Islands on macOS and you can't set Nagpur on Windows!). I didn't include the DST offsets as they are consistent.

| Location | GMT macOS  | GMT Windows
| :--- | :--- | :-- |
| New Zealand | -43200 | 43200
| California | -28800 | 28800
| Brussels | 3600 | -3600

Note: to get the New Zealand time, I applied the daylight savings offset of 3600.

It looks like the solution might a minus sign? Try this on line 663 in serial2xsbug:

```c
gmt = -tb.timezone * 60;
```

phoddie, 2020-12-08:
@wilberforce - anything?

wilberforce, 2020-12-11:
Sorry - Please see PR

closed 2020-12-21 by phoddie