Moddable-OpenSource/moddable #503: ESP32: TCP socket does not get closed completely in case of passive close

mlnuaa, 2020-11-25:
**Environment**

* Development Board: ESP32-DevkitC
* Module or chip used: ESP32D0WDQ6 (revision 1)
* ESP-IDF version: v3.3.2
* Moddable version: Commit (4fd7003)
* Build System: mccconfig
* Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
* Operating System: Windows7
* Using an IDE?: No
* Power Supply: USB

**Problem Description**
When a TCP socket close is initiated from the server with FIN flag data segment(i.e. Passive close for the client), the socket closure is not performing FIN exchange from the client-side. This makes the client TCP socket to remain in the CLOSE_WAIT state. The server also waits in the FIN-WAIT-2 state to receive the FIN message from the client.

**Expected Behavior**
The TCP passive close should follow the FIN exchange from the server and the client to ensure proper closure of the established socket as per [RFC793](https://tools.ietf.org/html/rfc793#page-39). 

**Actual Behavior**
In the case of TCP passive close, after receiving the FIN message, the client socket goes to `CLOSE_WAIT` state and not proceeding further to send the FIN message back to the server. The TCP state machine is waiting for the close from the application to send the FIN message to the server.

**Steps to reproduce**
The problem can be reproduced using the [socketreadwrite ](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/socket/socketreadwrite) moddable example application,
1. Configure verbose level of logs sdkconfig
2. Enable TCP debug options(TCP_INPUT_DEBUG, TCP_OUTPUT_DEBUG, TCPIP_DEBUG, LWIP_DEBUG, and TCP_DEBUG) in the lwipopts.h of esp-idf (Diff file attached) to get more details of TCP activities
3. Build and Run the socketreadwrite moddable example application in ESP32 with preconfigured network credentials and instrumentation enabled (mcconfig -i -m -p esp32 ssid=<network name> password=<password>)
4. Monitor the console logs 

**Code to reproduce this issue**
[socketreadwrite ](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/socket/socketreadwrite) moddable example application without any modification.

**Debug Logs**
```
[2020-11-25 11:11:14.595] IP address 192.168.137.179
[2020-11-25 11:11:14.598] tcpip_thread: API CALL message 0x3ffbfc80
[2020-11-25 11:11:14.602] tcpip_thread: API CALL message 0x3ffbfc80
[2020-11-25 11:11:14.605] tcp_bind: bind to port 62475
[2020-11-25 11:11:14.609] tcpip_thread: CALLBACK 0x3ffca088
[2020-11-25 11:11:14.612] socket created.
[2020-11-25 11:11:14.614] tcpip_thread: PACKET 0x3ffca088
[2020-11-25 11:11:14.617] tcpip_thread: PACKET 0x3ffce4dc
[2020-11-25 11:11:14.620] tcpip_thread: CALLBACK 0x3ffca088
[2020-11-25 11:11:14.623] tcp_connect to port 80
[2020-11-25 11:11:14.627] tcp_enqueue_flags: queueing 6509:6510 (0x2)
[2020-11-25 11:11:14.630] tcp_output_segment: 6509:6509
[2020-11-25 11:11:14.864] tcp_slowtmr: processing active pcb
[2020-11-25 11:11:14.870] tcp_slowtmr: polling application
[2020-11-25 11:11:14.875] tcp_output: nothing to send (0x0)
[2020-11-25 11:11:14.954] tcpip_thread: PACKET 0x3ffce660
[2020-11-25 11:11:14.959] TCP header:
[2020-11-25 11:11:14.963] +-------------------------------+
[2020-11-25 11:11:14.966] |       80      |    62475      | (src port, dest port)
[2020-11-25 11:11:14.970] +-------------------------------+
[2020-11-25 11:11:14.975] |           1419892360          | (seq no)
[2020-11-25 11:11:14.978] +-------------------------------+
[2020-11-25 11:11:14.982] |           0000006510          | (ack no)
[2020-11-25 11:11:14.985] +-------------------------------+
[2020-11-25 11:11:14.988] |  6 |   |010010|     65535     | (hdrlen, flags (SYN ACK 
[2020-11-25 11:11:14.992] ), win)
[2020-11-25 11:11:14.996] +-------------------------------+
[2020-11-25 11:11:14.996] |    0xdf1b     |         0     | (chksum, urgp)
[2020-11-25 11:11:15.000] +-------------------------------+
[2020-11-25 11:11:15.003] +-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags SYN ACK 
[2020-11-25 11:11:15.008] -+-+-+-+-+-+-+-+-+-+-+-+-+-+
[2020-11-25 11:11:15.012] State: SYN_SENT
[2020-11-25 11:11:15.015] tcp_parseopt: MSS
[2020-11-25 11:11:15.015] SYN-SENT: ackno 6510 pcb->snd_nxt 6510 unacked 6509
[2020-11-25 11:11:15.019] tcp_output: sending ACK for 1419892361
[2020-11-25 11:11:15.022] Socket message connectState: ESTABLISHED
[2020-11-25 11:11:15.027] 
[... 229 lines cut ...]
[2020-11-25 13:26:38.537] +-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags ACK 
[2020-11-25 13:26:38.539] -+-+-+-+-+-+-+-+-+-+-+-+-+-+
[2020-11-25 13:26:38.543] State: LAST_ACK
[2020-11-25 13:26:38.545] tcp_receive: ACK for 6571, unacked->seqno 6570:6571
[2020-11-25 13:26:38.548] tcp_receive: removing 6570:6571 from pcb->unacked
[2020-11-25 13:26:38.554] TCP connection closed: LAST_ACK 80 -> 57000.
[2020-11-25 13:26:38.557] tcp_pcb_purge
```

The Wireshark log also shows the exchange of `FIN` message from both the server and the client,
[image]

**Logs**
1. [Moddable_socketreadwrite_esp32_console_log.txt](https://github.com/Moddable-OpenSource/moddable/files/5597826/Moddable_socketreadwrite_esp32_console_log.txt)
2. [Moddable_socketreadwrite_esp32_wireshark_log.zip](https://github.com/Moddable-OpenSource/moddable/files/5597828/Moddable_socketreadwrite_esp32_wireshark_log.zip)
3. [Moddable_socketreadwrite_esp32_workaround_console_log.txt](https://github.com/Moddable-OpenSource/moddable/files/5597829/Moddable_socketreadwrite_esp32_workaround_console_log.txt)
4. [Moddable_socketreadwrite_esp32_workaround_wireshark_log.zip](https://github.com/Moddable-OpenSource/moddable/files/5597831/Moddable_socketreadwrite_esp32_workaround_wireshark_log.zip)

**Other items**
1. [sdkconfig.txt](https://github.com/Moddable-OpenSource/moddable/files/5597832/sdkconfig.1.txt)
2. [lwipopts.h.diff.txt](https://github.com/Moddable-OpenSource/moddable/files/5597834/lwipopts.h.diff.txt)

**TCP close references**
1. https://tools.ietf.org/html/rfc793#page-39
2. https://www.ibm.com/support/knowledgecenter/SSLTBW_2.1.0/com.ibm.zos.v2r1.hala001/closcispactive.htm

phoddie, 2020-12-07:
The lwip library used on ESP32 has two ways of reporting that a socket is going away:

- In the case of a remote close, it invokes the receive data callback with a null `pbuf`. 
- In all other cases, it invokes the error callback.

When the error callback is invoked, the lwip stack takes responsibility for freeing the socket. In fact, calling `tcp_close` can crash. On the other hand, in the case of a remote close, lwip requires the client to call `tcp_close` to terminate the socket. Until `tcp_close` is closed, the exchange with the remote is not completed.

For historic reasons, the ESP32 version of the code got this wrong and assumed that lwip would take care of cleanly closing sockets closed by the remote. The fix was simple -- removing the nulling out of the socket.

For those curious about why this is an issue, when the devices fails to acknowledge the close, the server sends it again. And again. If there are multiple unacknowledged closing sockets, the device can receive a flood of packets requesting to close sockets. This uses bandwidth and CPU. As a result, it slows all network operations.

There was a second bug. If there was unread data on the socket when the remote close was received, the JavaScript layer would not be notified that the socket is closed. This has also been fixed. This was a little tricky because the close notification  needs to be delivered after all of the received data has been read by the client.

mlnuaa, 2020-12-08:
@phoddie : I have verified with the commit ac3a6e8. The fix is working fine and closing the socket cleanly as expected. I think this issue can be closed.
Thanks for the quick resolution.

closed 2020-12-08 by phoddie