Moddable-OpenSource/moddable #507: How to use Moddable SDK with Platformio?

Ahmed-farag36, 2020-12-03:
I have checked Platformio docs and find how to program esp8266 with Arduino library _(c++)_ from **Visual Studio Code**, and it installed most of the Moddable SDK getting started requirements like toolchain-xtensa, esptool, sdk-esp8266 so it maybe work with Moddable SDK!

phoddie, 2020-12-08:
platformio has a very particular approach to building, one which does appear to be a good fit for Moddable SDK without significant customization work. I know. I tried. ;)

 If someone wants to  solve that problem, great.  But, it is not something we're going to tackle.

closed 2020-12-08 by phoddie

Ahmed-farag36, 2020-12-08:
Guide me and I may give it a try, what are the results of your try and what are the obstacles ?

phoddie, 2020-12-09:
I will try to explain my experience. Fair warning - I know very little  about platformio beyond my attempt to use it to build the Moddable SDK. if someone wants to explain that I got it all wrong, I will gladly agree. ;)

Platformio has a specific idea about how a source tree is organized. Independent of whether that is good or bad, it is different from the Moddable SDK. To do anything, I had to reorganize the Moddable SDK to match what platformio expects:

```
- sdkroot
 - include
 - lib
  - appscript
   - src
    - mc.defines.h
    - mc.esources.c
    - mc.xs.c
    - mc.xs.h
  - digital
   - include
    - modGPIO.h
   - src
    - modGPIO.c
  - i2c
   - include
    - modI2C.h
   - src
    - modI2C.c
  - resource
   - src
     - Resource.c
  - time
  - timer
  - xs
   - include
    - xs.h
    - xsmc.h
   - src
    - xsAll.c
    - xsAll.h
    - (etc)
   - xsPlatform.h
 - src
  - main.cpp
  - xsHost.c
 - test
```

The Moddable SDK is not just a native build. There are scripts and resources. Those are built with tools the platformio doesn't know about. To make that work at all, I manually ran xsc, xsl, etc. to generate the files that go into appscript/src That needs to be automated.

Any customization of platformio seems to require Python scripting. That's powerful, but since I'm no Python programming that's also impenetrable.

Eventually, it worked well enough to blink an LED and read a sensor or two.

But, I cannot imagine rearranging the entire SDK structure for one build system. There would need to be a more flexible solution. The more custom scripting that is needed for the build, the less obvious the benefit is of using platformio in the first place.

Ahmed-farag36, 2020-12-09:
Thank you so much for your fast response and all your efforts, may be I am
not understanding the system totally (being embedded systems beginner), I
will make some research and hope I can help.