Moddable-OpenSource/moddable #508: allow trailing .js in module specifiers

dckc, 2020-12-09:
We (@kriskowal @erights and co) have got code that uses explicit extensions in module specifiers:

```js
import x from './m1.js';
```

We're struggling to get this to work with `mcconfig` and `manifest.json`

**Why do you think this feature would be useful?**

Using ESM in node.js and the browser only seems to work with explict `.js`. @kriskowal and others can tell you the constraints in much more detail.

**Describe alternatives you've considered**

After much head-scratching and flailing with various manifest idioms, we're currently using an xs-specific branch with the `.js` removed: https://github.com/Agoric/SES-shim/commit/75eb85cccaf3ed5d6b4a6dcc741e430253e06b14

But we don't think this is feasible going forward.

phoddie, 2020-12-09:
Yes, agreed. This comes up often enough.  We need a supported solution.

The right way to solve it isn't entirely obvious. We'll take another run at it.

kriskowal, 2020-12-10:
To write code that can be reused between XS and Node.js, it might be possible to train Node.js to look for an implied `.js` with import maps, but that’s very new and unstable. I’ve yet to get it working. https://nodejs.org/api/packages.html#packages_subpath_imports

I’ll post here should it become unnecessary for our purposes for XS to do anything extra.

phoddie, 2020-12-10:
@kriskowal - yes, import maps could be one way of addressing this. They have the benefit of not requiring any Moddable SDK changes. ;)  But, I don't think they are widely available in browsers (I could be wrong... I really don't  work on the web!), so they probably aren't a general solution.

kriskowal, 2020-12-10:
`package.json` imports are related but not the same as import maps. It’s true that import maps are not widely deployed. The web currently only supports the explicit `.js`, so you are right that a sufficiently general solution might have to tolerate them.

kriskowal, 2020-12-10:
For our purposes at Agoric, for unrelated reasons, we’re pursuing a work-around that makes this issue non-urgent.

The unrelated reason is that our SES shim has a dependency on a CommonJS JavaScript parser. We would in time like to replace this, but it is not on the critical path at the moment.

The unrelated work-around is to use `compartment.evaluate` on a resource generated by Rollup, which covers all the Node.js package layout and module resolution wiggles and also the CommonJS dependency.

phoddie, 2020-12-10:
That's quite a workaround. ;) Glad you aren't blocked. We've got this in our queue.

phoddie, 2020-12-16:
We've integrated support for this in today's release. The `.js.` and .mjs` extensions are ignored. It works well in our tests. I suspect you have more complex scenarios in mind though, so we'll have to see if it works for all your cases.

kriskowal, 2020-12-16:
This is certainly sufficient for the subset of Node.js packages that do not contain any CommonJS modules.

phoddie, 2020-12-16:
Cool. When you can give it a try, we'll find out if we overlooked something.

closed 2021-01-15 by phoddie