Moddable-OpenSource/moddable #509: environment variables in "config" manifest section

dckc, 2020-12-09:
A `$(MODULES)/files/*` idiom works in the `"modules"` section, but it doesn't seem to work in the `"config"` section.

**Why do you think this feature would be useful?**

@kriskowal has his code checked out at `/Users/kris/ses` but mine is at `/home/connolly/agoric/SES-shim`; I'd like to access test files in the respective directory without editing the code.

**Describe alternatives you've considered**

command-line arguments: not available in the `mac` simulator platform
runtime environment variables: resorting to FFI for this seems kinda harsh
code generation - write a .js file and import it: could work I suppose

phoddie, 2020-12-09:
This was already raised in #468 and discussed there.

> command-line arguments: not available in the `mac` simulator platform

They are. For example, modify $MODDABLE/examples/helloworld/main.js to the following:

```js
import config from "mc/config"

trace(config.foo, "\n");
```

Then build and run:

```
cd $MODDABLE/examples/helloworld
mcconfig -d -m foo=bar
```

The output is:

```
bar
```

closed 2020-12-09 by phoddie

dckc, 2020-12-09:
ok. Thanks!

I know I looked at #468 at one point but I missed this detail.

Perhaps add a link from the [docs on the config section of the manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#config) to the [mcconfig CLI docs](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#mcconfig)? The latter suggests only 3 specific parameters can be assigned; it could be more clear that others are supported.

phoddie, 2020-12-09:
Good idea. I've made a few updates to the docs to try to explain that more clearly. Those will go live with the next repository update. Thank you.