Moddable-OpenSource/moddable #513: "Hello World" demo can't load simulator.so due to missing __atomic_fetch_sub_8 on Raspberry Pi

natevw, 2020-12-11:
**Build environment:** Linux (raspian "Buster", 5.4.51-v7l+ arm7l)
**Target device:** desktop simulator

**Description**

Trying to get the SDK going on a Raspberry Pi 400 with my sone, we're following the [Linux installation instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#lin-instructions) in the Getting Started guide works until the last step.

At that point, the instructions call to test the Hello World app via `cd $MODDABLE/examples/helloworld; mcconfig -d -m -p lin`. The compilation seems to work okay, but when it launches the GUI app we get an error dialog:

> Cannot open "/home/pi/.config/tech.moddable.simulator.so"
> /home/pi/.config/tech.moddable.simulator.so: undefined symbol: __atomic_fetch_sub_8

(I'm not sure if it's relevant to the underlying build/linking issue, but I should also note that often when this dialog comes up it's actually half-filled with random garbage characters instead! What it says above is the message when the output happens to be clean.)

This seems related to #301 but our checkout of this repo includes those fixes but is still giving this error in the simulator.

natevw, 2020-12-11:
I'll also note I tried running `mcconfig -d -m -p x-lin` since it looked like it was the "x-lin" makefiles that got updated rather than the "lin" ones. But when I do that the actual building fails with a bunch of other errors, so I'm assuming I'm not supposed to do that.

mkellner, 2020-12-11:
Hello. At line 46 in `build/makefiles/lin/simulator.mk` please try adding `-latomic` to the `LIBRARIES` assignment.
Please confirm if this works for you as I don't have a Pi handy to test this on.

phoddie, 2020-12-16:
@natevw  - did @mkellner's suggestion resolve your problem? Please let us know so we can update the repository, if appropriate, and close this out. Thank you.

closed 2020-12-22 by phoddie

natevw, 2020-12-23:
I think this needs to be re-opened. (Sorry, we had moved on to other projects in the meantime and didn't have a chance to test until this evening.)

I did the following:

```
cd $MODDABLE
git pull
cd examples/helloworld
mcconfig -d -m -p lin
```

One of the makefiles was updated with the pull, and the mcconfig command did appear to rebuild the helloworld again, but still results in the same garbled error as the UI part launches. (Just in case it was using some stale build products and not knowing the mcconfig equivalent of `make clean`, I also tested on another example which I'd never compiled before — it also gets the same error when launching the simulator.)

natevw, 2020-12-23:
Whoops, spoke too soon! This is probably fixed.

I had assumed only the example project itself would need rebuilding. When I rebuild the whole Moddable checkout (https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#lin-update) the missing symbols issue goes away. [Nothing appears on the simulator screen and I'm still troubleshooting that but regardless that would be a different issue than this.]