Moddable-OpenSource/moddable #515: Build option for runtime symbols of dynamically evaluated code

kriskowal, 2020-12-11:
**Describe**
Stack traces in XSBug name frames according to the function called, but for code run by `eval`, all such frames are marked `?`. Is there a way to build an XS executable that will show symbols for code evaluated at run-time?

[image]

**Why do you think this feature would be useful?**
At @agoric, we need a single executable that can execute arbitrary applications. With our current experiment, the application code is not inspectable in a debugger.

**Describe alternatives you've considered**
We’ve decided against compiling whole applications into unique executables since this would entrain the compiler in our trusted compute base, and because we have some CommonJS dependencies that are cost prohibitive to eliminate entirely, particularly the stand-alone version of the Babel parser framework. We also considered depending on Babel directly since it is ESM, but it entrains Node.js-specific built-in modules.

kriskowal, 2020-12-11:
Curious. I’ll attempt a tighter isolation.

kriskowal, 2020-12-11:
I was unable to reproduce named stack frames in isolation.

[image]

Platform: Mac OS X.15.7 (Catalina). Moddable base branch `9b2399f5` + changes to add TextEncoder and TextDecoder. The manifest is unremarkable except an oversized creation block. XS platform is default, `mac`.

phoddie, 2020-12-11:
Note that XS does have an option to exclude the `name` and `length` properties from functions. We use that to reduce flash size, since those properties are generally not needed. I'm not sure that is relevant here.

Something is different between our builds. My test run was run by replacing the source of $MODDABLE/examples/helloworld/main.js and then running the usual `mcconfig -d -m`. If you do the same, what is your result?

kriskowal, 2020-12-11:
I’ve taken the liberty of checking out `public` aka `9b2399f522f23d8fb812200e79ee99dd8d9f670f` and rebuilt Moddable tools.

Looks good in place!

[image]

If I delete our `creation` block in `manifest.json`, I see symbols in the stack trace! Looks like we just need to navigate to a creation block that has enough room for the run-time symbol table.

closed 2020-12-11 by kriskowal

phoddie, 2020-12-11:
@kriskowal - curious, but good. ;) If this issue makes a return, just let us know.

kriskowal, 2020-12-11:
Sure thing. I’ve yet to come up with a creation block that turns the ?’s into names for our non-trivial 3MB bootstrap source, so I’m going to spend some time getting to know creation block docs.

phoddie, 2020-12-11:
Yea, that's part of what I don't completely understand. If XS fails to allocate memory (for example, for a function name), it doesn't continue - it ends up in `fxAbort` and exit. It lands there if the symbol table is full too. So, perhaps there's something else going on here.

kriskowal, 2020-12-11:
For reference, this isn’t enough yet — mind that I’m blindly doubling numbers:

```json
	"creation": {
		"static": 32768,
		"chunk": {
			"initial": 1536,
			"incremental": 1024
		},
		"heap": {
			"initial": 1024,
			"incremental": 1024
		},
		"stack": 524288,
		"keys": {
			"available": 4194304,
			"name": 65536,
			"symbol": 65536
		},
		"parser": {
			"buffer": 4194304,
			"table": 4096
		},
		"main": "main"
	}
```

kriskowal, 2020-12-15:
I believe I need to submit a follow-up issue to capture the underlying problem, and I could use some help framing that issue.

I will be able to eventually share the branch, though I’m sure ideally I would whittle it down or binary search for the bundle size where function names turn to question marks in the debugger. The project branch is riddled with debts, as we try to spike to SES-tests-working-in-XS. The program I’m debugging is a 3MB bootstrapping bundle generated by Rollup, containing our SES shim and Node.js-package-graph to compartment-graph assembler.

I’m going to turn my attention to other debts for a while, but will eventually need to return to this debugging problem. My time is yours if I can help track this down. kris@agoric.com.

phoddie, 2020-12-16:
Your manifest numbers are kind of crazy. XS is written first for a microcontroller so there are some limits along the way. In particular, there is a hard limit on the number of symbols of 32767. Setting `keys.available` above that is unexpected. :) 

I was able to reproduce a problem similar to yours by adjusting `keys.available`. That value is actually the number of symbols available beyond what is built-in (JavaScript built-ins plus preloads). For hello world, I turned it down to 32000 and everything worked. Somewhere between there and 32767 things go bad.

I didn't need to change any other numbers from your creation record. That said, some of them are staggering (at least from my microcontroller-centric perspective). The stack, for example, is measured in slots not bytes, so your stack is 8 MB on a 32-bit system (and more on a 64-bit system). But, that seems to work.

Anyway... please try adjusting `keys.available` down. It would be good to know if that resolves the problem. We can look at some better error reporting by the tools once we are sure of the problem. Thanks.

phoddie, 2020-12-16:
I should also mention that `keys.name` is used for hashing the name table. It should be a prime number. It doesn't need to be particularly large. It impacts performance and memory use, but it isn't any kind of functional limit. We use 53 for microcontrollers. For larger systems we use 1993, which is what I'd suggest here.

kriskowal, 2020-12-16:
Thanks for the details. I’ll try these recommendations out!

kriskowal, 2020-12-17:
I did get another symbol in my stack trace with these creation block details.

[image]

phoddie, 2020-12-17:
Curious. Do those `?` functions have a real name or are they anonymous functions? Also, I see `async` in there. Is any of that pending async functions?

kriskowal, 2020-12-17:
Ah, indeed. Digging around in the artifact generated by Rollup, there are in fact three IIFE’s before and after createCommonjsModule. So, it would seem this configuration 100% solves the function naming problem.

It would seem that the level up from here would be stack frame line numbers and, if present, the file location from a trailing `//#sourceURL=bootstrap-stage-2.js` comment. I can request that feature separately.

phoddie, 2020-12-17:
OK, great. One down. ;)