Moddable-OpenSource/moddable #516: http put (Server.requestFragment) does not receive complete file

wilberforce, 2020-12-14:
**Build environment:** Windows
**Target device:**  windows/esp32

**Description**
If the file put is larger than the input buffer, the final bytes are not sent.

**Steps to Reproduce**
1. https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpserverputfile/main.js
2. `mcconfig -d -m -p win`
2. via curl send a file that is larger than the buffer size, in my case I'm sending www.xsa, which is 2399
`curl -X PUT -H "Content-Type:application/octet-stream" --data-binary @C:\Users\rhys\Projects\moddable\build\bin\esp32\debug\www\www.xsa http://192.168.69.80:80/www.xsa -v`
3. The saved file is 2*1024  = 2048, there are still 351 bytes left that are not read and written.

**Expected behavior**
The callback is called for the last 351 bytes.

**Other information**
The issue appears to be around this logic:
https://github.com/Moddable-OpenSource/moddable/blob/ece6a84f3190d1191bf4698c656aad0755d6d5e3/modules/network/http/http.js#L611-L621

First pass:
count=1024, value=1024, this.total=2399
count=351, value=1024, this.total=0

I modified the sample so the file is written to the same folder:
`this.file = new File('./'+path, true);`

phoddie, 2020-12-14:
Good find.

It looks like the problem is that the received bytes are double counted when the callback requests `Server.requestFragment` messages. That got broken by other changes.

The change below resolves the double counting:

```

wilberforce, 2020-12-14:
@phoddie - thanks
That work around appears to work.
 > It does, however, expose another problem with reporting at the end of the request. I'll look into that later.
I'm not sure how this manifests?

wilberforce, 2020-12-23:
Working now - thanks!

closed 2020-12-23 by wilberforce