Moddable-OpenSource/moddable #517: Error trying to use mcconfig with simple typesscript project.

ariznaf, 2020-12-15:
I have tried to use typescript in a simple project that just imports couple of simple modules with one functiona and import Timer from timer module.

When I try to run mcconfig I get an error complaining about timer not being found, after running tsc tsconfig.json to generate the tsconfig.json for tsc to run properly.

The example is here (typescript test):

https://github.com/ariznaf/pruebasMoodable/tree/main/typescriptTest

ariznaf, 2020-12-15:
After investigating the problem for a while, I could find a possible solution.
It seems that the generated tsconfig.json lacks some path to the directory that contains xs modules.

I have added this line to the path section:
```
  "*": ["D:/Users/fernando/ProyectosElectronica/moddable/typings/*"],
```

And then when you run tsc (in the tmp directory) it runs with no complains.

But when you run mcconfig again, the tsconfig is overriden and you get the error again.

This is the complete tsconfig.json file for the example that seems to work with tsc:

```
{
	"compilerOptions": {
		"baseUrl": "./",
		"module": "es2020",
		"outDir": "d:\\users\\fernando\\ProyectosElectronica\\moddable\\build\\tmp\\esp32\\debug\\typescriptTest\\modules",
		"paths": {
		        "*": ["D:/Users/fernando/ProyectosElectronica/moddable/typings/*"],

			"main": [
				"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\test\\main"
			],
			"util/color/blue": [
				"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\lib\\color\\blue"
			],
			"util/color/red": [
				"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\lib\\color\\red"
			]
		},
		"sourceMap": true,
		"target": "ES2020",
		"types": [
			"d:\\users\\fernando\\ProyectosElectronica\\moddable\\xs\\includes\\xs"
		]
	},
	"files": [
		"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\test\\main.ts",
		"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\lib\\color\\blue.ts",
		"D:\\Users\\fernando\\ProyectosElectronica\\MisProyectos\\pruebasModdable\\typescriptTest\\lib\\color\\red.ts"
	]
}
```

phoddie, 2020-12-15:
To use the contents of $MODDABLE/typings it is necessary to include those in the project. The easiest way to do that is to include `manifest_typings.json` in your project's manifest. For example:

```
{
    "include": [
        "$(MODDABLE)/examples/manifest_base.json",
        "$(MODDABLE)/examples/manifest_typings.json"
    ],
    "modules": {
        "*": "./main"
    },
}
```

This is explained, along with other helpful information, in our blog post [(Very Experimental) TypeScript Support in the Moddable SDK](https://blog.moddable.com/blog/typescript/).

closed 2020-12-15 by phoddie