Moddable-OpenSource/moddable #518: It's very troublesome to set up the environment in Windows system.

zhuangbo214, 2020-12-16:
It's very troublesome to set up the environment in Windows system. I set up the environment on two computers. One of them can't work, and I can't find the reason.

Whether you can develop an installation program for beginners, just click Install step by step.

Or it would be better to have a plug-in on VS Code.

phoddie, 2020-12-16:
@zhuangbo214, sorry to hear that you are having problems.

Setting up for native embedded development is can be challenging, especially when getting started. We would like to help. If you can share the problem you are seeing, for example the error message you receive, perhaps we can help.

We regularly update our set-up instructions based on feedback from developers. If something is unclear, please let us know the specific text that you did not understand.

Thank you.

phoddie, 2020-12-21:
Closing this issue out because of a lack of information and activity. If you'd like us to help with this is the set-up issue, please re-open this with the details of the problem. Thank you.

closed 2020-12-21 by phoddie