Moddable-OpenSource/moddable #519: XSBug: Stack frame line numbers and file names for `eval` and `compartment.evaluate`.

kriskowal, 2020-12-17:
**Describe**
Currently, XSBug labels stack frames for dynamically evaluated code with only the function name (or `?` for anonymous functions). It would be helpful to see line numbers at least, and maybe even reveal the corresponding source code on the right. Some files may have a `//#sourceURL=` pragma at the end of the file. https://developers.google.com/web/updates/2013/06/sourceMappingURL-and-sourceURL-syntax-changed This would allow in some cases for the debugger to show the alleged file name for the evaluated code.

**Why do you think this feature would be useful?**
This would be useful for debugging dynamic code, for example programs generated by Rollup or Agoric’s nested evaluator mode for packaged contracts.

**Describe alternatives you've considered**
We considered building a binary for every Agoric contract, strictly using ESM for contracts and their transitive dependencies. Disallowing CommonJS is very limiting but also survivable. However, building a binary for every contract would require us to execute build tooling outside the object capability paradigm, which would be inimical to our design, requiring us to fall back to Docker containers and build proofs, or some such.

kriskowal, 2020-12-17:
Following #515

phoddie, 2020-12-17:
I spent some time reviewing the XS sources this morning to understand what is possible. I'd like to understand your needs a bit better to see if there's a useful subset of those which are practical to implement.

The `//$sourceURL=` pragma is new to me. I believe XS could be coaxed into supporting that. It is a little convoluted because the parser reads the source text as an input stream from start to finish. Because the sourceURL is at the end, the parser would need to update the references after generating the parse tree.

What looks like an unrealistic effort is getting  in-line string literals to work, e.g. `eval("function foo() {debugger;} foo();")`. But, it doesn't sound like that's needed here?

If the source text is loaded from a file using the host, the everything works as you want already. I don't suppose that's practical here because you are applying transformations to the code before evaluating it?

kriskowal, 2020-12-18:
Perhaps the stream order problem could be solved with a level of indirection, filling the AST with a pointer the the pointer you’ll fill in at the end of the stream. Though, it might be sufficient for the AST to refer to an object that contains the source. The sourceURL pragma is an affordance that Firebug introduced and has since become a norm for all developer tools, but not a norm of the language. The debugger usually gleans the alleged sourceURL after the fact. For the SES-shim, I’m extracting it with a regex only if we throw an error that needs file and line number.

Supporting the sourceURL pragma is completely unnecessary except in the case of evaluating a string literal with either `eval` or `compartment.evaluate`. We can certainly stuff as much of our bootstrapping code as possible into the XS binary with compiled modules, but contracts will be introduced to xsnap as either strings containing JavaScript [[Program]]s (in the short term) or byte buffers containing Zip files containing ESM [[Module]]s and CommonJS module files (in the distant term).

After we’ve cleared some near-term hurdles, we intend to visit dynamic module loading first by running SES-shim on XS, then getting the Compartment-as-module-loader spec in order based on that experience to invite the possibility of an XS implementation with feature parity sufficient to express Node.js applications with both ESM and CJS.

We may be able to live without being able to debug contracts on XS, provided that we also build a Node.js contract runner that with identical behavior. To do that, we’ll have to find a way to get the bootstrapping code into an XS compiled module. Having talked through that 🐥, it should be possible to trick Rollup into generating a [[Module]] instead of a [[Program]].

phoddie, 2020-12-18:
Thanks for the background. That helps.

It sounds like having sourceURL would be useful, though it isn't mission critical for tomorrow. I believe that is possible to support. Given that it is a widely supported hack, we can try. ;)

phoddie, 2020-12-21:
@kriskowal - good news. Patrick had an insight for how to implement something more general than sourceIRL. We did some work on that over the weekend. This screen recording shows the result.

[image]

We'll look at getting that pushed in for you later today.

kriskowal, 2020-12-21:
This will help materially! We’ll be able to refer the sourceURL at the
bottom of the eval code visually to find the original material too.
Surfacing the eval code in the debugger at the line number of the stack
frame solves most of the immediate problem!

phoddie, 2020-12-21:
Cool! This work gets the fundamental mechanism in place for debugging code evaluated at runtime. We think we can refine this further with some additional effort.

What's particularly nifty, at least from my perspective, is that this works for `eval` on a microcontroller too (thought we still don't recommend doing that...).

phoddie, 2021-01-15:
The fundamental issue reported has been resolved, so closing this out. If there are specific details to address with the current functionality, please open a separate issue.

closed 2021-01-15 by phoddie